SimplifiedLogic/creoson #2: UNICODE support

YaroslavSin, 2017-09-27:
I'm playing with Playground and found that the search by parameters not working with names of parameter in cyrillic. The same problem with string value in cyrillic.

adama2000, 2017-09-28:
Hm, can you give us a test case?  Such as a sample part and a test application which shows the problem?

YaroslavSin, 2017-10-04:
https://cloud.mail.ru/public/AKXD/kayyR7RTN

Part for testing.
Parameters:
TEST_RU_VALUE = Строка текста на русском
ТЕСТ_РУ_ЗНАЧЕНИЕ = Test ru text
ПРОВЕРКА = значение провеки

adama2000, 2017-10-04:
Thanks, we're working on the issue.

adama2000, 2017-10-05:
Fix is posted to the source repository, and included in the 2.1.0 release just published.  This was not a Playground issue but an issue with the core code which was not dealing with Unicode characters properly.

closed 2017-10-05 by adama2000