SimplifiedLogic/creoson #5: Retrieve model from Windchill

FrostTiger, 2018-02-01:
I have dealt with creoson with creating http post request using postman for 2 days and I like it very much and very nice tool. But I couldn't retrive a model from windchill server. Is it capable to do retrieving model from windchill server? 

When I check J-Link user guide that PTC provided. The command has already exist in pfcSession.BaseSession.RetrieveModel 

I have check the source code as far as I understand. just I wanted to be sure whether the function was not implemented from your side.

adama2000, 2018-02-01:
Creoson does use the RetrieveModel function in its "file : open" function.  That is normally used to open a model from the hard drive or a network drive, but can also open files from WindChill if you first use the "windchill" functions to authorize (login), set a server, and set a workspace.  

At least that's how other users have used the software to get files from WindChill.  Note that file-open will not check out a file from WindChill, but it will extract a copy of the file into the workspace.

FrostTiger, 2018-02-02:
After setting server. It works very well. thanks

closed 2018-02-02 by FrostTiger