SimplifiedLogic/creoson #7: Interface : mapkey function

DG1987, 2018-07-11:
I'm having issue's with the interface : mapkey function.
It works to play the first line, but nothing happens for the lines on the next command.

It is also possible to play it line by line, but the end result is slow...

Can somebody  check if they are having the same problem?

Is there something missing in the explanation to add the next line:
{
  "sessionId": "~sessionId~",
  "command": "interface",
  "function": "mapkey",
  "data": {
    "script": "~ Select `main_dlg_cur` `View:casc340798662`;mapkey(continued) ~ Close `main_dlg_cur` `View:casc340798662`;mapkey(continued) ~ Command `ProCmdNamedViewsGalSelect`  `FRONT`;"
  }
}

adama2000, 2018-07-11:
Try taking out the "mapkey(continued)" parts from the middle, see if that helps.  When using the "interface:mapkey" function, the tildes should come right after the previous semicolon, like this:

"script":"~ Select "main_dlg_cur" "View:casc340798662";~ Close "main_dlg_cur" "View:casc340798662";~ Command "ProCmdNamedViewsGalSelect" "FRONT";"

DG1987, 2018-07-12:
Perfect, It works if I remove "mapkey(continued)"
I will contact documentation with a request to change it.

closed 2018-07-12 by DG1987

adama2000, 2018-07-12:
Checked in a doc change for interface:mapkey to warn users about the issue.