SimplifiedLogic/creoson #8: Adding Material File

DG1987, 2018-07-17:
I would propose a new function for Creoson:

file : Set_Material
=> add a Material for a file, with an option to also activate the material
(activating the material is also possible by changing parameter [PTC_MATERIAL_NAME])

dmtr99, 2019-08-13:
I managed to get a working code to modify the material, but I have no idea hot to insert this in CreoSon:

		// Set Material of part.
		ModelType modeltype = model.GetType();
		System.out.println("CurrentModel: " + model.GetType());
		if (modeltype.equals(ModelType.MDL_PART)) {
			System.out.println("Model is part");
			 
			fileName = fileName.replace(".mtl","");
			File file = new File(fileName);
			session.ChangeDirectory(file.getParentFile().getAbsolutePath());
	    		// Retrieve the material item.
			Material newmaterial = ((Part) model).RetrieveMaterial(file.getName());
			((Part) model).SetCurrentMaterial(newmaterial);
			System.out.println("Changed Material of part: " + fileName);
			return "true";
		}

adama2000, 2019-09-15:
Thanks for that.  I've checked in code changes to add some new functions for handling materials.  The new functions will be included in the next release.

file : load_material_file
file : set_cur_material
file : get_cur_material
file : list_materials
file : delete_material

dmtr99, 2019-09-16:
Thank a lot,
Looking forward for the next release :)

closed 2019-09-17 by adama2000

dmtr99, 2019-10-22:
Just Tested the new release functions (dimension set, material set, material load, dimension list).
It works!
Thanks a lot.