SimplifiedLogic/creoson #10: Rename through CREOSON

DieSwartKat, 2019-06-18:
I'm trying to rename a file in CREO using CREOSON. I'm sending this:

{'command': 'file', 'data': {'new_name': 'prt0017a.prt', 'file': 'prt0017.prt', 'onlysession': True}, 'functions': 'rename', 'sessionId': '8048605998689615145'}

then I get this:
({'status': {'error': False}}, False)

No renaming happened. I'm having trouble seeing what I'm doing wrong.

adama2000, 2019-06-18:
I tested your input string, and it worked for me and renamed the file.  However, I also got more output data.  What you SHOULD have gotten back was:
{"status":{"error":false},"data":{"file":"prt0017a.prt"}}

Looking at the source code, I can see two reasons why you might not get "data" output --
1) Your session has expired or you're passing an invalid sessionId, or
2) The file and new_name arguments match (meaning the file has already been renamed)

It looks like the second one isn't true, so could your session have expired?  The creoson server looks the sessionId up in its internal list of active sessions.  If it can't find the sessionId in memory, then the function will return with no output.

DieSwartKat, 2019-06-18:
Thanks for the response.

I'm using Python to send the Json to the server. The check for the server and getting the session ID is build into the python code. The code check if CREO is open and if the session is alive and print the session id every time.

Here is the communication:
"""Checking if CREO is running"""
{'function': 'is_creo_running', 'command': 'connection'}
Creo is active
api url:http://localhost:9056/creoson
self:{'WORK_DIR': 'C:\\\\\\\\PTC\\\\\\\\Bar_Start_Creo', 'CONNECTING_RETRIES': '0', 'API_URL': 'http://localhost:9056/creoson', 'START_DIR': 'C:\\\\\\\\Program Files\\\\\\\\PTC', 'START_FILE_NAME': 'nitro_proe_remote.bat'}
getting session id ...
{'function': 'connect', 'command': 'connection'}
session id:   2964917986368891547 False
 """Setting the working directory"""
{'function': 'cd', 'data': {'dirname': 'C:\\Users\\mcronje\\Documents\\00_Work\\1701622_Eucerin_Full_Drop-NEW_LOOK_2017'}, 'command': 'creo', 'sessionId': '2964917986368891547'}
{'function': 'open', 'data': {'files': 'prt0017.prt', 'dirname': 'C:\\Users\\mcronje\\Documents\\00_Work\\1701622_Eucerin_Full_Drop-NEW_LOOK_2017', 'display': True, 'new_window': True, 'regen_force': True, 'activate': True}, 'command': 'file', 'sessionId': '2964917986368891547'}
prt0017.prt
{'functions': 'rename', 'data': {'new_name': 'prt0017a.prt', 'file': 'prt0017.prt', 'onlysession': True}, 'command': 'file', 'sessionId': '2964917986368891547'}
({'status': {'error': False}}, False)

I've also tried other names than prt0017a.prt but to no avail. 

As you can see I'm bringing the file into session, but when I'm trying to rename the change doesn't happen. I've also checked the file in session:

[image]

It is still the same name.

DieSwartKat, 2019-06-18:
I've also checked the source code to see if I can find the problem. The only time I expect a "null" return from Jlink will be if newFilename is not nothing.
But I cannot see where I'm causing this to happen.

String newFilename = fileHandler.rename(filename, newname, onlysession, sessionId);
		
        if (newFilename!=null) {
			Hashtable<String, Object> out = new Hashtable<String, Object>();
       		out.put(OUTPUT_MODEL, newFilename);
        	return out;
        }
        return null;
	}

adama2000, 2019-06-18:
Ohhh... I just noticed your main problem.  Your JSON on that call contains "functions" rather than "function".  Because creoson doesn't see a "function" argument it just returns null.

Sorry I missed that earlier, I had written my own json rather than just copying yours.

adama2000, 2019-06-18:
I'll make a code change in the next build so you'll get an error message if that happens again.

DieSwartKat, 2019-06-19:
I'm slightly embarrassed, thanks so much! An return error message will really help, thanks for the help and prompt response.

davidhbigelow, 2019-06-19:
Ha - no worries, we ALL make that mistake at some point...  Glad it is
working!

adama2000, 2019-06-27:
We've just released 2.4.0, and this includes the error message I mentioned above.

DieSwartKat, 2019-06-27:
Thanks so much! You guys are doing amazing work.

DieSwartKat, 2019-07-05:
Just some feedback - we have build a localhost Django project where we interact with CREO by using CREOSON. It works like a charm and we are most definitely building more features using this amazing tool.

Thanks for all the help and support and for a great product.

closed 2019-07-05 by DieSwartKat

davidhbigelow, 2019-07-05:
WOW... That sounds really cool!

Do you have any examples you can share about how that works?  And/Or how
you are using it?