SimplifiedLogic/creoson #11: CREO Unable to connect to J-Link

DieSwartKat, 2019-06-26:
Good day

I'm trying to read the current working directory through the function "creo : pwd" It returns an error saying : 
'message': 'Unable to connect to Creo through J-Link', 'error': True

I can fix the error by restarting CREO, but then I get another error

This time is_creo_running returns  "status": {"error": true },  "data": {"running": false }

The command currently running after this is the creo:pwd command which returns the correct working directory even though it told me that CREO is not running.

In all the above the I've first run the command creo:connect which in all cases returned a new session ID and "error": False.

Since I'm not passing a session_id into is_creo_running (the documentation says it is ignored) does it mean that it could be looking at the wrong session and return an error?

kind regards

DieSwartKat, 2019-06-26:
I'm having trouble replicating my own error. The only thing I could find was that there seems to be a delay in communication between CREO Jlink and CREOSON. In some cases the communication doesn't return from CREOSON and some communication error occur. I'm fixing this by putting the JSON request in a "try" function. I'm having positive results from it. 

If I manage to replicate the problem I'll update this post.

adama2000, 2019-06-26:
If you think it's a speed issue, you can increase the connection timeout.  Find the .bat file "creoson_run.bat" and edit it.  Near the bottom is the command line for running the server.  On that command line is the option

`-Dsli.jlink.timeout=200`

That specifies that the connection timeout is 200 milliseconds.  You could try to increase that if the problem happens again.

**Note:** This assumes you're using creoson_run.bat to run the server, or that you're running the server from the Creoson Setup application.  If you're not, then you need to modify whatever script you are using instead.

The connection timeout doesn't apply to the is_creo_running function -- that has a fixed timeout of 10 milliseconds.

davidhbigelow, 2019-06-26:
Also - just to clarify ... The session_id is only required to be obtained
once - at the connect.

And then that session_id is reused for all subsequent calls made to
CREOSON.  Sometimes people think they need to re-connect before every call.

I realize you are NOT probably doing that - but wanted to make sure the
note about this was made in the feed for anyone else who searches for a
similar issue.

DieSwartKat, 2019-06-27:
Thanks for these comments, I am only using a single connection and reusing the session id. 
In my script I try to connect to CREOSON and if it fails I try to start CREOSON assuming it is not running. Sometimes it doesn't respond and the creoson_run.bat says "Address already in use: bind" which I assume means CREOSON server has stop responding.
This seems to happen if I put my machine in sleep mode and wake it up again.

Would it be possible to "force restart" CREOSON server, would that even solve the problem?

davidhbigelow, 2019-06-27:
Ah - "Sleep mode".  Not sure why "sleep mode" on your machine would cause
that.... but it may be that the "sleep mode" is resetting the Licensing for
CREO and the API interface can't re-connect to it initially on "awake" of
the machine.  JLINK does not require a license, but the PTC API's sometimes
get a block if there is any hic-up in the PTC Licenses Server (Sleep Mode"
can cause this).  We have seen that in the past (long time ago) with other
applications that talk to Pro/ENGINEER/Wildfire/Creo.

I think that this is not our issue... if your app can detect the problem
after your machine "wakes up"... I would recommend your external app bounce
the CREOSON server, with the CREOSON bat file that starts the server, when
it encounters this specific condition - then wait a few seconds for the
CREOSON server to come up - then try to transact again.

I would give that a try.

DieSwartKat, 2019-06-27:
Thanks I've done what you suggested and it works very well!

closed 2019-06-27 by DieSwartKat