SimplifiedLogic/creoson #12: Drawing : GetFormatPageList

DieSwartKat, 2019-07-05:
Would it be possible to build a function in CREOSON that returns the format used in a drawing per page? I need to identify what drawings are used for, so that I can name them accordingly. I cannot use the active part/assembly as we often have more than one drawing for the same part/assembly but with different formats. The only way to distinguish between these drawings would be to identify which format is used.

I'm open to suggestion should you feel there is an alternative way to solve this problem.

Kind regards

davidhbigelow, 2019-07-05:
What about the *drawing : get_sheet_size* as a possible workaround?

DieSwartKat, 2019-07-05:
> What about the *drawing : get_sheet_size* as a possible workaround?
> […](#)
> -- David Bigelow, President Simplified Logic, Inc <https://www.simplifiedlogic.com> | SimilarParts.ai C: 317-431-5454

Thanks good suggestion, but the get_sheet_size command only return the size (A4, A3, etc) of the layout and not the name of the format used (*.frm). That will not be enough of a distinguisher to help identify what the drawing is used for.

davidhbigelow, 2019-07-05:
Hmmm.. it looks like it is possible in JLINK... but like anything else - we
would have to verify that it works before we implement a change.

But as a possible (short term) workaround... I would look for a model
parameter that is typically associated with the type of drawing you are
looking to identify (e.g. plastic part vs sheet metal vs weldment etc
etc...)  and use that if you can in your logic.  This assumes corporate
standards are established and the part/assembly is more descriptive than
the drawing format used as to "what it is"

phaleth, 2019-07-06:
The following might help, it uses fmt setter.

```
Window window = session.GetCurrentWindow();
Model model = window.GetModel();
if (model.GetType() != ModelType.MDL_DRAWING) {
	throw new RuntimeException("Current model is not a drawing.");
}

Model2D drwModel = (Model2D) model;
String formatFileNameInclExt = formatFileName + ".frm";
int currentSheetNum = drwModel.GetCurrentSheetNumber();

ModelDescriptor modelDescr = pfcModel.ModelDescriptor_CreateFromFileName(formatFileNameInclExt);
RetrieveModelOptions rmo = pfcSession.RetrieveModelOptions_Create();
rmo.SetAskUserAboutReps(false);

DrawingFormat format = null;
try {
	format = (DrawingFormat) session.RetrieveModelWithOpts(modelDescr, rmo);
} catch (jxthrowable x) {
	throw new RuntimeException("Failed to retrive the format into session: " + x);
}

drwModel.SetSheetFormat(currentSheetNum, format, null, null);
drwModel.UpdateTables();
```

davidhbigelow, 2019-07-06:
Wow!   Thanks!!!!

Feel free to contribute more functions and ideas!!!

adama2000, 2019-09-17:
I've added two new functions to get and set the sheet format -- the get function will return the format file name and full name.  They'll be included in the next release.

closed 2019-09-17 by adama2000

DieSwartKat, 2019-10-14:
You Guys ROCK! Thanks for including get_sheet_format in the new CREOSON 2.5 release. We are implementing this as of today.

I love the fact that it returns both the FORMAT and the TEMPLATE used, this gives us extra functionality to make sure we keep the formats of drawings up to date.

Is there any place, or method where we can show what is possible with CREOSON. I'm considering opening a YouTube channel, but would like to reference Simplified Logics for the great work you are doing on CREOSON?

davidhbigelow, 2019-10-14:
HA - thanks it is a team effort + community + customer feedback that makes this better.

I love the idea of a YouTube Channel for contributions like this... We have a CREOSON YouTube Playlist  under our Simplified Logic (NitroMation) YouTube Channel that we have been adding things as we find them:

https://www.youtube.com/playlist?list=PLcWd1Ea7i_twWMA6NjkrfLsYpeNVQ9anw

Youtube has a weird policy on creating specific named sites - I would be open to that if it i makes sense.

It is a **good idea**... If you want to publish on your site and let us know you have done it, we can add it to our playlists so that people have a better chance of finding example and use cases.

What do you think?