SimplifiedLogic/creoson #13: Tables in drawings

dmtr99, 2019-08-13:
It would be interested to read and modify tables in a drawing:
- Get List of tables
- Read content of table
- Get position/size of table
- Create or insert new table
- Add new row or column
- Modify contents of a table

A use case for this would be for example a revision table.

adama2000, 2019-09-16:
We've looked at this in the past, and it's very complicated to cover all the bases properly with the limited functionality PTC provides beyond some basic creation / manipulation.

Low priority for us right now, but if someone wants to propose some code for inclusion we'll take a look at it.

closed 2019-12-09 by davidhbigelow

dummifiedme, 2022-12-10:
Was there ever any progress on the table in drawing front?
Is it possible to add a table into the drawing sheet using creoson?

dmtr99, 2022-12-10:
I did not found a good way to modify existing tables.

Creating is quite easy, check the format of the drawing and use a generated mapkey.
I would recommend to rename the table so you are able to search for it.

dummifiedme, 2022-12-10:
@dmtr99 thanks for the insight. I didn't know about mapkeys.
Please allow me to explain what I needed to do, and if you feel you can suggest how to proceed, please do.

I have a set of symbols for which I need to generate a table if they are on a sheet with their respective locations.
A column each for symbol, location and the dimension next to it.

I figured out everything using creoson/creopyson, but unable to find a way to put it into a table.

As you suggested using MapKeys, are these mapkeys capable to run a python script? As in, I would want to create a table using the CreoButtons I suppose which will maybe work using mapkeys (?) and then put in the values I retrieve using the python script. Is it possible?

Should I use mapkeys for compiling the whole thing and remove creoson altogether?

Any suggestion would really be helpful :)

dummifiedme, 2022-12-29:
@dmtr99 Can you please suggest how to add a table?