SimplifiedLogic/creoson #15: Improved set dimension

dmtr99, 2019-08-13:
Is it possible to improved the set dimension function with the ability to set the Prefix, Suffix and Text?

adama2000, 2019-09-17:
It appears PTC's API doesn't handle the prefix/suffix separately; instead, the prefix and suffix are merged in with the text (even though they aren't displayed that way inside the Creo Dimension Text editor).  So if you wanted to set the Prefix or Suffix, you'd have to just include them in the Text.

As for the Text, JLink expects you to send an array of text lines for dimension text; that text can contain dimension references, plain text, and symbols.  Symbols are the tricky part, because they are stored in binary form inside the text; for example, the plus/minus symbol is stored as hex(01)+"#"+hex(02).  Since JSON doesn't handle binary data, we'd have to either require that dimension text be Base64-encoded before sending, or provide some alternate way of specifying symbols that would get translated on our end, such as "{PLUSMINUS}".  What do you think?

adama2000, 2019-09-17:
Sorry for not responding earlier than this, we've been busy here.

dmtr99, 2019-09-17:
Seems logical.
If the function get dimension text exist, users always can figure out what the symbol is and this way you do not need to add any extra documentation.

adama2000, 2019-09-17:
Does that mean you think we should just require the user to specify dimension text in Base64?

adama2000, 2019-09-17:
Actually it's probably better to just handle it the way we do in note:set -- have the user set an 'encoded' flag if the value is base64-encoded.

dmtr99, 2019-09-18:
Correct, the function should be similar to other functions, this makes it simpler.

adama2000, 2019-09-18:
checked in code for a new dimension : set_text function

closed 2019-09-18 by adama2000