SimplifiedLogic/creoson #17: pdf Export tools 'use pentable'

anjansimkhada, 2019-08-20:
The export_pdf function is very fast compared to my current method of using mapkeys. 
Any chance of adding 'use pentable' checkbox when using the 'export_pdf' function?

davidhbigelow, 2019-08-20:
You can do this already with a few config.pro settings --- no need for a
special option on the command to force this behavior.

There are a couple of options.... I recommend this video as a starting
point:

https://www.youtube.com/watch?v=reyRAqXVnKI

And these for reviewing the configuration details:

http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/fundamentals%2Ffundamentals%2Ffund_two_sub%2FExample_table_pnt_File.html%23
http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/fundamentals%2Ffundamentals%2Ffund_two_sub%2FSpecifying_Location_of_Pen_Table_File.html%23

If you want for "force specific settings" for pen tables (e.g. different
pen table file) - you can do this by setting configuration options on the
fly in your program using the creo : set_config command and options.

Hope that helps!

Dave

anjansimkhada, 2019-08-20:
Hi Dave,

Thanks for that.
I'm trying to export pdf for a huge top level assembly drawings (>500 parts) using export_pdf.
The current way of doing it is by running a trail file that does open>save a copy>pdf, etc. to the specified directory continuously.

However, the 'export_pdf' function is much faster than the trail file method. 
I've setup my pentable.pnt file and pointed the config option to the right directory as well.
I've also changed the use pentable config option to 'yes'.

Still the 'export_pdf' seems to save the pdf without using the pentable.
If you've got any suggestions, I'open to them. Thanks!

davidhbigelow, 2019-08-20:
Couple of things that seemed to make a difference for me:

note full path to pentable file:
pen_table_file c:\working\pentable.pnt

and these options also seem to be required:
pdf_use_pentable yes
use_8_plotter_pens yes

And this was the test pentable.pnt I was messing around with:
---- START pentable.pnt ----
pen 1 thickness 0.0013 in; color 0.0 0.0 0.7
pen 2 thickness 0.0013 in; color 0.0 0.7 0.0
pen 3 thickness 0.0013 in; color 0.7 0.0 0.0
pen 4 thickness 0.0013 in; color 0.1 0.0 0.0
pen 5 thickness 0.03 in; color 0.0 0.2 0.0
pen 6 thickness 0.0013 in; color 0.0 0.0 0.3
pen 7 thickness 0.003 in; color 0.4 0.0 0.0
pen 8 thickness 0.0013 in; color 0.0 0.5 0.0
---- END pentable.pnt ----

If you change the thickness value from 0.0013 ---> 0.13 -- you should see
changes in your exports.

Then it is just trial and error to map out the pen numbers to the proper
export settings you want.

Hope that helps!

Dave

anjansimkhada, 2019-08-20:
awesome. I just needed to point the pen_table_file to the full path. Earlier it was just pointing to the pentable.pnt file. Thanks for your help. Much appreciated. I can't wait to test out the rest of the tools. Cheers!

closed 2019-08-20 by curiosity-rover