SimplifiedLogic/creoson #18: Searching through all the server side workspaces

curiosity-rover, 2019-08-21:
Currently, I can search through the active LOCAL workspace using 'windchill_list_workspace_files'.
Is it possible to search through the SERVER workspace aka folder contents (which lists all the files within that workspace in the server)?

davidhbigelow, 2019-08-21:
I don't believe this is possible via the JLINK APIs... We will have to
double check that...

phaleth, 2019-08-21:
With the <code>wtpub://</code> prefix on the path it is possible to list things located in Commonspace.

The J-Link example bellow goes through all folders in a Windchill Library. Notice that the getListOfSubDirs() method is recursive.

```
/**
 * 
 * @throws jxthrowable 
 */
@Override
public void OnCommand() throws jxthrowable {

	session = pfcGlobal.GetProESession();

	Server server = session.GetServerByAlias("Windchill");
	if (server == null) {
		session.UIShowMessageDialog("Could not retrieve the Windchill server", null);
		throw new RuntimeException("Could not retrieve the Windchill server");
	}
	
	String alias = server.GetAlias();
	server.Activate();

	String workspace = server.GetActiveWorkspace();
	if (!workspace.equals("Workspace on NORMALIZOVANE_DILY")) {
		server.SetActiveWorkspace("Workspace on NORMALIZOVANE_DILY");
	}

	String wChPath = "wtpub://Windchill/Libraries/NORMALIZOVANE_DILY";
	
	List<String> subDirs = getListOfSubDirs(session, wChPath);
	
	SimpleDateFormat tf = new SimpleDateFormat("yyyy-MM-dd HH:mm:ss");
	
	for (int i = 0; i < subDirs.size(); i++) {
		String subDir = subDirs.get(i);
		stringseq fileFullPathsSeq = listLatestFiles(session, subDir);
		for (int j = 0; j < fileFullPathsSeq.getarraysize(); j++) {
			File fileFullPathObj = new File(fileFullPathsSeq.get(j));
			String fileName = fileFullPathObj.getName();
			
			seq.set(0, "[SubDir " + (i + 1) + "/" + subDirs.size() + "] "
					+ "[Model " + (j + 1) + "/" + fileFullPathsSeq.getarraysize() + ": " + fileName + "]);
			session.UIDisplayMessage(MSG_FILE, "msg.infoMsg", seq);
			seq.clear();

		}
	}
	
	session.UIShowMessageDialog("Done.", null);

}

/**
 * 
 * @param session
 * @param dir
 * @return 
 * @throws jxthrowable 
 */
public static List getListOfSubDirs(Session session, String dir) throws jxthrowable {
	
	List<String> list = new ArrayList<>();

	stringseq dirPathsSeq = listSubDirs(session, dir);
	for (int i = 0; i < dirPathsSeq.getarraysize(); i++) {
		File dirPathObj = new File(dirPathsSeq.get(i));
		String dirPath = dirPathObj.getPath().replace(":\\", ":\\\\").replace("\\", "/");
		list.add(dirPath);
		List<String> subList = getListOfSubDirs(session, dirPath);
		list.addAll(subList);
	}
	
	return list;
	
}

/**
 * List only the latest version of all files in a specified directory
 *
 * @param session
 * @param dir
 * @return 
 * @throws com.ptc.cipjava.jxthrowable 
 */
public static stringseq listLatestFiles(Session session
		, String dir) throws jxthrowable {
	
	stringseq fileFullPathsSeq = session.ListFiles("*.*"
			, FileListOpt.FILE_LIST_LATEST, dir);
	return fileFullPathsSeq;
	
}

/**
 * List subdirectories of a specified directory
 * 
 * @param session
 * @param dir
 * @return 
 * @throws com.ptc.cipjava.jxthrowable 
 */
public static stringseq listSubDirs(Session session
		, String dir) throws jxthrowable {
	
	stringseq fileFullPathsSeq = session.ListSubdirectories(dir);
	return fileFullPathsSeq;
	
}
```

adama2000, 2019-09-19:
@curiosity-rover Can you clarify?  It sounds like you have a workspace X, and you want to be able to list the contents of the server version of workspace X because the local version of workspace X doesn't have all the files in it.  I thought the two workspaces were supposed to be kept in sync, at least if you're working online.

curiosity-rover, 2019-09-20:
@adama2000 SERVER workspace has all the files but the local workspace only lists the files that are new  or have been retrieved/added to workspace. So, say my server workspace has 10 assemblies with 10 unique parts each, so 100 files in total. When I open 1 assembly that will only add 10 parts to my local workspace, the rest 90 won't be added as they are not called by that specific assembly.

Just for scale, one of my server workspace has got 40,000 objects (.asm, .prt, .drw ,etc). But, when I want to open a single assembly of a mere 50 parts, I don't need to clog my local workspace with all those 40,000 objects. This keeps my local workspace neat and doesn't slow down Creo. Hope this is clear.

phaleth, 2019-09-20:
Commonspace a.k.a server-side workspace a.k.a Windchill Cabinets (or whatever PTC calls it inside Creo Parametric nowadays) is accessed from J-Link with the `wtpub://` prefix.
One of user's workspaces a.k.a. client-side workspace a.k.a Workspace (or whatever PTC calls it inside Creo Parametric) is accessed from J-Link with the `wtws://` prefix.

The only solution I know of is that all the filepaths need to be pre-loaded into the Java runtime, as shown in the above code.

Be aware that pre-loading of all the filepaths with J-Link that exist in a product/library might be lenghty and also unreliable, depending on how many subfolders has to be retrieved.

weston-ta, 2019-09-20:
@curiosity-rover, Good Day.
In reading through this thread I am a bit confused as to the issue at hand. Please forgive me for asking for an explanation of the full issue you are faced with as it sounds like the discussion has moved from one sub-issue to another.
Allow me to ramble a bit. At the heart of PDMLink is the server. All interaction with it is done through a web page, or one that is embedded inside of Creo.
You must create an use the workspace to deposit items into the server. Once "checked-in" those files are safe and the workspace can and should be deleted. In the example above that mentions the 40000 objects. If they have all been checked in, OR at least the subset of it (the 50) that you want to work on then you can create a brand new "Clean" workspace, attach your creo session to it and continue on your way to designing. Once you are done with those 50 you can check them back in and "update" the workspace with the 40000 to see the effects of the changes there.

If as the title suggests, "Searching through all the server side workspaces" you have quite a number of workspaces and the something is hindering you from "Check-in" then the discussion is a much longer one.

curiosity-rover, 2019-09-24:
@weston-ta Yes the check-in process is correct. But, currently Creoson only allows you can to search through your local workspace using 'windchill_list_workspace_files'. So, say I've got 1000 parts with prefix 1testxxx.prt (where xxx=0 to 1000). I create 1001, check-in, delete my workspace. Now, I've got a total of 1001 parts  of interest. Now, these 1001 parts can be scattered across different server workspace (in my case, around 10 different server workspaces). This happens because every user might work on any one of these 10 workspaces and they check-in the parts in that workspace, hence, scattering these parts across multiple workspaces.

Currently, the only way of listing these 1001 parts is by adding them ALL to my local workspace. I wanted to be able to search these 1001 parts directly from the server workspace (or commonspace or Windchill Cabinet as @phaleth has mentioned above). S/He is also correct about the prefixes regarding the local and serverside workspaces.

'windchill_list_workspace_files', currently, only works if that part is in your local active workspace. It doesn't look into server side workspace.  Hope this clarifies things a bit.

PS:
Server side workspace OR Windchill commonspace OR Windchill cabinet is accessed by wtpub:// prefix
Local/client workspace OR simply just workspace is accessed by wtws:// prefix

closed 2026-08-20 by davidhbigelow