SimplifiedLogic/creoson #19: Erase part from workspace

curiosity-rover, 2019-08-25:
Is it possible to Erase the file from the current workspace?
I know there's windchill_clear_workspace that clears the entire workspace, wondering if we can pass "file_=" to select an specific one.

curiosity-rover, 2019-08-27:
> Hm, there is a modelNames option on the JLink call we're using, but we're not exposing that in the creoson interface. That could be a change.

That'd be fantastic if it gets added, it allows us to search through the server side Windchill workspace without clogging up the current workspace. Search for part(>1000 objects)>>read params>>close and clear from workspace, was my initial motivation for this post.

adama2000, 2019-09-19:
Checked in code for this, it'll be in the next release.

closed 2019-09-19 by adama2000