SimplifiedLogic/creoson #22: BOM: get_paths

DieSwartKat, 2019-09-04:
We have been playing with the bom:get_paths function. There are several frustrating things about it that causes us to question the structure of the json which it returns. I'm specifically talking about  the default settings for this function (not top-level)

Because assemblies are all unique the structure of the json returned is different every time. As a result finding information in the json is very difficult.

We would like to suggest a different approach that rather uses the component as the key and the seq_path as the value.

Because of the smart way in which the seq_path is communicated you will clearly see where each element is and which components are children and where there parent are.

{ main_asm.asm : root ,
sub_asm01.asm : root.1,
part01.prt: root.1.1 ,
part02.prt: root.1.2
sub_asm02.asm: root.2,
part03.prt: root.2.1 }

As a user I'm going to asking questions like:
1. Where is prt01.prt?  The answer is root.1.1
2. What is at root.1.1? answer: part01.prt
3. What are the children of sub_asm01.asm?  answer: [part01.prt, part02.prt]
4. What is at root? Answer: main_asm.asm

Unless I'm missing something there seems to be no benefit in the json code following the same structure as the model being interrogated by CREOSON.

Please share your thoughts.

davidhbigelow, 2019-09-04:
We are providing the same structure as the model...  which is the basis for
ALL BOM structures, and variants, that anyone might want to derive.

The structure returned should be returning the same every time - for the
same assembly.

The structure you are suggesting could be easily built in your client using
the CREOSON BOM as an input.  OR you could seek a specialized library (eg
graph structure) to boost your understanding of the relationships you are
seeking.

I don’t think we will be changing the current function unless there is a
genuine bug in the current reporting.  What you are asking is a bit too
specialized for this information.

Dave

DieSwartKat, 2019-09-05:
Thanks David we did manage to get the Json parsed to the format we wanted. We are using Python, here is the code should anyone else have the problem.

`import json

def find_key_value_pairs(q, keys, dicts=None):

    if not dicts:
        dicts = [q]
        q = [q]  

    data = q.pop(0)
    if isinstance(data, dict):
        data = data.values()

    for d in data:
        dtype = type(d)
        if dtype is dict or dtype is list:
            q.append(d)
            if dtype is dict:
                dicts.append(d)

    if q:
        return find_key_value_pairs(q, keys, dicts)

    return [(k, v) for d in dicts for k, v in d.items() if k in keys]

main=json.loads(data)[0]["data"]["children"]
keys = ['seq_path', 'file']
results = find_key_value_pairs(main, keys)
`

First we convert the json to a dictionary and remove the top level as it doesn't have the seq_path key in it.

The variable "results" is an array which I referred to in the top question.

closed 2019-09-05 by DieSwartKat

davidhbigelow, 2019-09-05:
AWESOME -- and thanks for posting some code back to the community on how to
get a re-formatted version.

There are a LOT of ways to format BOM data for multiple use cases and
scenarios.

I am glad you found CREOSON useful for your goals and might help others
with similar concerns.

Well done!

Dave