SimplifiedLogic/creoson #24: Creoson Visual Studio

bharathhsb7, 2019-09-13:
Hi,
I would like to know how to use Creoson Library into Visual Studio for VB.
Thanks

davidhbigelow, 2019-09-13:
Language specific interfaces are typically required to talk to the HTTP
server.  I can only tell you to look for an HTTP request protocol within
your chosen language.

Some languages are easier than others for this type of integration (and
easier to use) than VB.  I have seen some people using VB with CREOSON -
the code looks incredibly heavy with "quotation" management, python or node
might be a lot easier to work with and code.

Dave

dmtr99, 2019-09-16:
Autohotkey is also a possibility.

Here is a link to a working code:
https://community.ptc.com/t5/Detailing-MBD-MBE/CREOSON-Beta-3-is-Available/td-p/475608/page/2

closed 2019-09-19 by davidhbigelow