SimplifiedLogic/creoson #25: Search tool - selection

dmtr99, 2019-09-18:
I currently have a beta script that analyses the program of a part and generates a drawing.

When the views are placed, it uses "show model annotations" and it needs to select whitch dimensions needs to be visible.

For this, it uses tha mapkey function to find and select certain ID's.
This works, but is quite slow if it needs to find 50 dimensions.

Is there a way to select certain dimensions through J-Link?

It would be interesting to have a function that acts the same as the search tool, so you can add objects to the selection. This would open a lot op possibilities.

phaleth, 2019-09-18:
Linking a thread where are some partial answers as to how to do selection of drawing items and drawing tables with J-Link code.

https://community.ptc.com/t5/Customization/How-to-get-the-value-of-BOM-balloon-index-text-available-in-the/m-p/34739

DetailTypes are specified in J-Link API documentation jlinkdoc as follows

[image]

I guess dimensions are detail items of DetailType.DETAIL_NOTE and these can be casted to Dimension2D object type, but not sure, never tried that.

Also there is a sparing use of SelectionBuffer shown in one of the examples.

adama2000, 2019-09-18:
JLink does have a way to programmatically select items.  It can also prompt the user to manually select items.

There is already a Creoson function -- dimension : user_select.  This prompts the user to select a given number of dimensions, and when the user clicks OK in Creo the function returns with a list of the dimensions selected.

On the other hand, if you want dimensions to be selected programmatically rather than by the user, I imagine there's something that could be done.  Do you just need to get a list of dimensions that match certain criteria?

How do you think any proposed Creoson function would look as far as inputs and outputs?

dmtr99, 2019-09-19:
if possible, I would like to make is similar and flexible like the search tool function, this would be quite powerfull and flexible.
Maybe something like this:
{
  "sessionId": "~sessionId~",
  "command": "selection",
  "function": "add_search",
  "data": {
    "type": "dimension",
    "search": "name",
    "searchvalue": "d1",
    "clear_selection": true
  }
}
type could be feature, dimension, drawing table, drawing note, drawing symbol, drawing view, datum plane, datum axis, ...
search could be id, number, name, all
clear_selection could be true or false
The output could be the amounth of selections found and the total amounth of the selection

another function list could maybe give you an array of the selected objects, (type, name, id)

But I assume that it is simpler to make it more specific to dimensions:
command: dimension
function: add_selection
request:
- search: id, number, name, all
- clear_selection: true or false
- searchvalue: the value to search

a second command to check wat is selected:
command: dimension
function: list_selection
response could be similar to "dimension : list"

adama2000, 2019-09-19:
So the point is not to get a list of dimensions (which we already do) but to leave Creo in a state where certain dimensions (based on the search) are in a selected state?

dmtr99, 2019-09-20:
That is correct, in combination with the mapkey function, you can simulate a lot of user commands.
For example:
-  removing dimensions on a drawing
- adding dimensions on a drawing
- modifiïng the tolerance of a dimension

if this function would not only work for dimensions, but more general like the search tool, it could also be used to remove tables, notes,...
It is possible to use the mapkey command to command the search tool, but this is rather slow and the script cannot see what is found or in the selection.

adama2000, 2019-09-20:
I've spent some time looking at it, and tinkering with test JLink code, and it appears to me it's not possible.

The reason is that while you can use JLink to programmatically select items, that is an internal selection and has no bearing on what the Creo window **thinks** is selected.  That would be fine if you wanted to do something with the selection internally in JLink, but running a Mapkey afterwards would probably fail because the Creo window doesn't think anything is selected.

JLink does allow you to highlight the selected items so that they **look** selected in the Creo window, but that doesn't appear to actually put them in a selected state as far as the window is concerned.

Whether there's another way to do it is a question for PTC, I'll submit a question to them about it.

adama2000, 2019-09-20:
Wait, maybe I'm wrong.  Let me read more about Selection Buffers...

adama2000, 2019-09-20:
Or maybe just add a flag to the existing list functions to select the items that are found.

adama2000, 2019-10-09:
I've checked in code which adds a 'select' parameter to the list functions under note and dimension, so that will be in the next release.

closed 2019-10-10 by adama2000