SimplifiedLogic/creoson #26: feature - resume with feat_id

Zepmanbc, 2019-10-01:
Hello,

is there a way to resume a feature with the "feat_id"

Something like:
```
{
    'command':'feature',
    'function':'resume',
    'sessionId':'123456789123456789',
    'data':{
        'file':'myfile.prt',
        'feat_id':'12345'
    }
}
```

davidhbigelow, 2019-10-01:
I don’t think so...  we set that up for name-based to keep it simpler for
automation.

Can you try to rename the feature then suppress/resume?

I am sure it would be possible - but we felt that it would be really hard
to require IDs.

Dave

Zepmanbc, 2019-10-01:
Hi Dave,

I want to resume one by one every feature, I get all the features with *features/list* so I get all ordered IDs.
I was thinking the same, I can rename all unnamed features with nearly the same automatic name because with the result of feature/list I got the feature type and the ID so it will be the same but capitalized
automatic name: *Plane id12345* => named feature: *PLANE_ID12345*

DieSwartKat, 2019-10-02:
We have done something similar with use of mapkeys.
Create a search qry file upfront using the find command (binocular icon). Make sure to include the Feat ID in the search CRITERIA. We also use the Feat # . We pass the ID and # into the mapkey on a loop. Then you can find/select the feature and then run a mapkey to unsuppreess it. 

It's a little bit like playing darts, but with some care it works 99% of the time.

Zepmanbc, 2019-10-02:
I renamed the unnamed features, but i got a proglem with GRAPH that i could not rename, and so it did not regenerate.
But if I put all theses unnamed features and "impossible to rename features" in a named group, it works fine 👍

adama2000, 2019-10-11:
I've checked in code to let you suppress and resume with a feature id.  It'll be in the next release.

closed 2019-10-11 by adama2000