SimplifiedLogic/creoson #27: More than one Distinct Pieces

DieSwartKat, 2019-10-02:
Good day everyone!

Is there a way to find out if a part has more than one distinct pieces. I have not been able to use the mapkey functionality to find something something useful.

If it is not clear - more than one distinct piece refer to Multi Body parts (Solidworks). There are two solid bodies in a part model.

kind regards

davidhbigelow, 2019-10-02:
List feature types?

davidhbigelow, 2019-10-02:
> Hi David
>
> I'm a little confused as how the feature list will see that the part has
> distinct pieces. None of the parameter information in features show this
> information. Am I understanding you wrong or missing something?

I was thinking that the "two solid bodies" in your part are special feature
type in Creo (e.g. from import or creation by operation).

If they show up in the feature tree (creo graphical window) - then the
"type" of feature should be available if you do a feature list to see what
the type is for each feature.

Assuming that the features you are interested in are of the same type...
then you just have to feature list using a type filter and you should be
able to count them up generically.

Unless I am missing your goal.  Just a suggestion on how to possibly obtain
the information.

DieSwartKat, 2019-10-02:
I have not seen that information in the feature list, will do a few more test to see if I can get it.

Currently CREO creates a little icon called distinct pieces on the bottom right of the screen. It is active if I make a extrusion with two loose sketch which makes two solid bodies. The search for solid body only show one, but distinct pieces show more.

One will not be able to create the Flatpattern feature on a part with distinct pieces, which kind of mess with our system. If would like to isolate it from other parts that have missing flatpattern features so that I can create drawings in the right format.

Let me investigate the feature list further.

DieSwartKat, 2019-10-03:
http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/sheetmetal/sheetmetaldesign/Your_Design/Working_with_Two_or_More_Distinct_Pieces.html

This is what I was refering too, if there is still any unclarity. Features List don't have the information I'm looking for.

davidhbigelow, 2019-10-03:
Oh...  Hmmm that is a bit more complicated than I thought you were dealing
with.

Do you have a specific model you can share that exhibits this issue along
with some instructions as to what you are looking for?

Dave

closed 2019-12-09 by davidhbigelow