SimplifiedLogic/creoson #28: Automate the CAD Conversion

neonaik, 2019-10-18:
Hi 
Is there any way that we can import cad files from another software and convert them in creo automatically with the help of creoson ? manual conversion is taking time as well as loosing the features.

any insights will be really helpful.

MFG
Ramesh

adama2000, 2019-10-18:
Only if 
1) The conversion is supported by Creo itself, meaning if you can do it in Creo by importing the cad file and saving it manually, and
2) The conversion is supported by JLink functions (I can research that).

On the other hand, if you're thinking that creoson could do a conversion by parsing the file itself rather than going through Creo -- that's outside the scope of creoson, which is just intended to be a convenient wrapper for Creo functionality.

Like I say, I can look at whether the API can support what would essentially being opening a cad file and doing a "save as" -- but if you're losing features when you do that manually, then this method will probably lose features too.

neonaik, 2019-10-18:
Hi at first thank you for your reply

In manual conversion the user is loosing the sketches with which the user
did some operations for ex padding but while converting the model exists
but lost its base profile

And thank you for your interest if you could find some API's please do let
me know

Mfg
Ramesh

adama2000, 2019-10-22:
I found out that the API has an ImportNewModel() function which can be used to import certain file types into Creo.  After that, just doing a save saves the model as Creo model.

It only supports certain file types.  What is the file extension for the file you import?  I can check to see if it's supported.

Note that even if we get this to work, it may well STILL lose the sketches.  This is basically just doing an Import and a Save.

davidhbigelow, 2019-10-22:
I would be shocked if PTC allowed features to be imported (e.g. including
sketch references and constraints from another system) ... I think that is
what the Associative Topology Bus (ATB) is supposed to do to keep things in
sync.  But this I think also assumes the Flexible Modeling extension for
modification.... Not sure if that is an option on that ImportNewModel()
call...

tliebenow, 2019-10-22:
I suspect that function is related to the Unite functionality.
-t

closed 2019-12-09 by davidhbigelow