SimplifiedLogic/creoson #29: Feature: Make File Read-Only or Have windchill like Lock

calederer, 2019-10-22:
Is it possible to have a function to make a file read only/modifiable or have a "windchill lock" without a windchill?

adama2000, 2019-10-22:
In theory, we could add a function to make a file read only/modifiable if it's on the local system.  Whether we _should_ or not is another question.

I don't know what you mean by a windchill lock without windchill.

adama2000, 2019-10-22:
Hm, I see by your separate email I misunderstood what you meant.

calederer, 2019-10-22:
Yup. Unfortunately Pro/E ignores the read only flags set by the files system. 

Unless i'm missing something I think PTC leaves a read only function out of the API so your it's harder for PDM companies to interface with it and force you to buy windchill.

adama2000, 2019-10-22:
It makes sense that WF5 ignores the fact that a file is marked "read-only" in the operating system -- because WF never modifies the original file. it just creates a new file with a new revision number.

Zepmanbc, 2019-10-22:
I'm interested for a read only or continue option for Windchill if it is possible
I've made a function with autopygui to click on the continue option but it is dirty :confounded:

calederer, 2019-10-22:
@Zepmanbc - I believe windchill should do this out of the box. or are you looking for a windchill like function for working with files in a directory?

Zepmanbc, 2019-10-22:
I'm opening assemblies from Windchill and want to automatise screenshots of different views.
I want to supress some parts depending the view for the screenshots so Windchill ask me to check out, but i don't want to. (It is legacy assemblies so there is no simplified views i should need)
That's why i'm interesting for a Windchill function to set Read-only or Continue when i open the assembly.
something like this maybe:
(status is maybe not the good word, but i'm not fluent in english)
```
{
    'command':'windchill',
    'function':'status',
    'sessionId':'123456789123456789',
    'data':{
        'file':'myfile.prt',
        'status':'continue'
    }
}

davidhbigelow, 2019-10-22:
Would a Mapkey be a possible workaround?  (assuming there is a manual
option for this?)

calederer, 2019-10-22:
Windchill already does this. You can use this set all function for the assembly

[image]

davidhbigelow, 2019-10-22:
Hmm - is there a config.pro option for that?

calederer, 2019-10-22:
Not sure: https://www.ptc.com/en/support/article/CS96637

Zepmanbc, 2019-10-22:
Yes but I have to do it manually for each assembly
I got a list of assembly and lanch the script

I've tried the mapkey but it does not work, this Windchill window seems to "freeze" Creo

I did not tried config pro, i will have a look

davidhbigelow, 2019-10-22:
Yes - sorry the mapkey did not work - some PTC modules don't really embrace
that.... unfortunately.

calederer, 2019-10-22:
From the Jlink API

The method pfcWPP.WPPServer.LockServerObjects
establishes an explicit lock on the specified objects on the server.
Specify the full path, name, and extension for the input objects.
This method returns the pfcServerLockConflict object that
contains the details of conflicts, if any, that occurred during the lock
operation. Use the method
pfcWPP.ServerLockConflict.GetObjectName to access the
name of the object for which the lock conflict occurred. Use the
method pfcWPP.ServerLockConflict.GetConflictMessage to
get details of the lock conflict.
The method pfcWPP.WPPServer.GetServerObjectLockStatus
checks the lock status of the specified object on the Windchill
ProductPoint server. Specify the full path, name, and extension for
the input object. The method returns the pfcServerLockStat
object that contains information regarding the lock status.
The method
pfcWPP.WPPServer.GetServerObjectsLockStatus checks the
lock status of a set of objects on the Windchill ProductPoint server.
Specify the full path, name, and extension for the input objects. The
method returns an array of pfcServerLockStat objects that
contain information regarding the lock status of the input objects.
Use the method pfcWPP.ServerLockStat.GetObjectName to
access the name of the object, including the extension, for which the
lock status is described.
Use the method pfcWPP.ServerLockStat.GetStatus to access
the status of the lock on the object on the server.
• PRO_OBJ_LOCK_STAT_UNSET—Specifies that no lock has
been applied on the object.
• PRO_OBJ_LOCK_STAT_HARDLOCK—Specifies that objects
are locked and that the current user is not the owner of the
locks. Therefore, this user cannot modify or release the lock.
• PRO_OBJ_LOCK_STAT_SOFTLOCK—Specifies that the
objects are locked and the current user is the owner of the
locks. Therefore, this user can release the lock.
• PRO_OBJ_LOCK_STAT_UNLOCKED—Specifies that the
explicit or implicit lock has been removed from the object and it
is available for editing.
Use the method pfcWPP.ServerLockStat.GetStatusMessage to
access the status of the object on the server. It provides the name of
the user who locked the object and the time of locking.
The method pfcWPP.WPPServer.UnlockServerObjects unlocks
a set of objects that have been explicitly locked on the product
server. This method returns the pfcServerLockConflict object
that contains the details of conflicts, if any, that occurred during
the unlock operation.

adama2000, 2019-10-22:
That might be worth looking at.  However, The pfcWPP functions were removed in later versions of Creo; Creo 2.0 / WF5 is the last version where the pfcWPP package still exists.

We're thinking we don't want to add functionality that only works on really old versions of Creo/WF and won't work on newer versions.

adama2000, 2019-10-22:
If there _is_ a config.pro option that does what you want, then you could still use the existing creoson function to set that option after you open a model.

closed 2020-02-26 by davidhbigelow