SimplifiedLogic/creoson #30: Windchill: check out, check in, Revise and check out

dmtr99, 2019-10-23:
Is it possible to Command Windchill even more?
- windchill : check_out                            file, files
- windchill : check_in                              file, files, context 
- windchill : revise_and_check_out         file, files

@adama2000 : your work is mindblowing, it is amazing how many possibilities for improvement and automation it opens.

davidhbigelow, 2019-10-23:
@dmtr99 .... Thanks for the praise... it is appreciated! :)

Some of that appears to be possible.  We just do **not** have a local installation of Windchill to test.   We tried to set that up once a few years back and put our hands in our face and stepped back...

Not sure if the setup/installation has gotten any easier or not... We have been blessed to work with a customer or two that enables us to validate some basic things with our APIs into it.

Will have to noodle on that a bit more.  A bit fearful of executing some of that functionality on a customer's environment (cringe)...  Maybe we attempt another installation marathon session at some point - just got to find the time...

Zepmanbc, 2020-03-26:
I'm also interested with theses features, maybe i can help for tests?

bvandeursen, 2024-01-08:
After playing around a bit with Creoson (I'm impressed, well done!), I fully expected these functions to be implemented. To me, and I expect many PTC users, being able to check in/out of Windchill is crucial for any automation script. 

Any idea if this is going to be implemented? Also willing to help with any necessary testing!

thackb, 2024-11-15:
I'm interested with these features as well, Would be good to be able check out / check in on demand.
I can also help with testing.

RMT-ME-AB, 2025-08-11:
This functionality is partially available already by running mapkeys tied to the check-in/check-out features under the creo parametric file tab. We just built some of this internally with a simple python script running creopyson and the 4 mapkeys: checkin, checkout, undo_checkout, and revise_checkout. If you combine revise_checkout followed by undo_checkout you can get a pure revise (although there is no way to set the revision so you are stuck with the next rev index). The limiting issues here are that I believe the mapkeys only work on the active file so you would need to loop through a list if you wanted to modify a batch. Secondly creo will complain if you try to save a change before the checkout is complete so I have been doing the following to cope with that:

-Open file
-Run checkout mapkey
-While loop checking for windchill file checkout out to return True
-Modify part
-Save part
-Run checkin mapkey

closed 2026-08-20 by davidhbigelow