SimplifiedLogic/creoson #32: select feature

dmtr99, 2019-10-24:
Is it possible to get and set selection information?
- I want to know which feature or features are currently selected
- And have the possiblility to add another feature to the selection (or deselect the current feature and select another.)

Use cases:
1. When making holes, I prefer to start from a sketch of points to use it for a point pattern. This is usefull because there is a lot of flexibility and you can define symmetrich pattern quite easily. The only disadvantage is if you want to modify quickly the position of the holes. With these function you could:
- let the user select a hole
- let the user activate a script with a hotkey or menu
- the script wil read what is selected, select the parent sketch and use the Edit or Edit definition function.

2. When modifiïng a Extrusion, it is faster to select the internal sketch and do a edit definition of thies (and by this skipping the extrusion tab that is not needed), a script could identify the sketch, select it and use the edit definition mapkey

adama2000, 2019-10-24:
Yes, it's possible to get a list of the current selections, we'd just have to decide what information the creoson function should return.

It's possible to add things to the current selection, but specifying the items to add can be tricky.  Adding components or model items (including features) to a selection is fairly easy, as long as we have the name or ID of the item.  Adding other things (for instance, drawing views) to a selection can be trickier because it involves some funny syntax.

dmtr99, 2019-10-27:
In the function dimension : list_detail, there is a parameter select to select dimensions.

So maybe it would be logical to add something to feature : list

Another idea is to manipulate the search tool of creo or have a function that behaves is the same way but does not displays the search window, this would be powerfull and extremely flexible to select a wide range of different objects. In the Search tool there is also a list of the current selected Items. If we could read that list we would have at least some information on the selection.

Haschtl, 2020-10-06:
Is the feature_select function still in development?

I currently use ugly mapkeys to select features in the model, which does not work stable in all cases.
Just specifying the path of KE-IDs to the feature would be really nice.

adama2000, 2021-02-22:
I've added a feature.list_selected function.  

I did work on a function to set a selection -- however there seems to be a bug on PTC's side where I can make it select features in the top-level assembly, but if I try to make it select features in a sub-component it selects one in the top-level assembly that has the same Feature ID.

Haschtl, 2021-02-23:
Thank you very much! 
the list_selected feature is already very helpful to check if my mapkey (to select parts using the KE-IDs) actually selects the correct parts. 

isn't the "select_feature" function some kind of recursive function? e.g. given a list of KE-IDs [1,2,3], we need to first find the KE-ID 1 in the topmost layer, then search for KE-ID 2 inside Part 1, and so on.

adama2000, 2021-02-23:
The way it's SUPPOSED to work is that you send the JLink function the
feature object and the component path to the feature.  But it seems to be
ignoring the component path and instead selecting the corresponding feature
ID in the active model instead.

dmtr99, 2021-02-24:
Thanks, this opens again a lot of possibilities.
I'm looking forward to test it in the next release!

closed 2021-03-21 by adama2000