SimplifiedLogic/creoson #33: Add line to message area

dmtr99, 2019-10-24:
Is it possible to write a log line to the message area to indicase something was done by the script to the user?
This would give a professional vibe to the script.

Getting the status of the notification icon would also be usefull, I do not like the small notification, you often notice late that an error occured like the saving of a part failed.

adama2000, 2019-10-24:
In theory; there's a PTC API function to display a message in the message area.  One restriction is that messages have to come from a file in the "text" folder under Creo's "Common Files" folder, so part of your setup would have to be setting up that file in your Creo installation.  Then you would tell Creoson the name of the message file, the line number in the file for the message, and a list of arguments to replace any parameters in the message.

I don't think there's anything in the API for getting the notification icon status.

YaroslavSin, 2019-11-14:
In text file you can create empty message like this
empmes
%0s
#
# 

Then print any text from run-time using this message.

Messages can have icon status

%CImessage1
This is information message
#
#

%CWmessage2
This is warning message
#
#

And so on. Take a look at documentation about this.

closed 2026-08-20 by davidhbigelow