SimplifiedLogic/creoson #35: How to get *node number* for Mapkeys?

Zepmanbc, 2019-11-04:
Hello

I'm importing STL file in a part and want to select a Csys for postionning it.
I have done it with a mapkey
```
"~ Command `ProCmdModelImpAppendNew` ;\
~ Trail `UI Desktop` `UI Desktop` `DLG_PREVIEW_POST` `file_open`;\
~ Trail `UI Desktop` `UI Desktop` `PREVIEW_POPUP_TIMER` \
`file_open:Ph_list.Filelist:<NULL>`;\
~ Select `file_open` `Ph_list.Filelist` 1 \
`my_stl_file_to_import.stl`;\                                   <= HERE IS THE STL FILE NAME
~ Command `ProFileSelPushOpen_Import@context_dlg_open_cmd` ;\
~ Activate `intffeat_file_dlg` `Dlg_PushBtn_Ok`;\
~ Activate `main_dlg_cur` `chkbn.intffeat_reference.0` 1;\
~ Select `intffeat_reference.0.0` `PH.CsysRG` 1 `Csys`;\
~ Select `main_dlg_cur` `PHTLeft.AssyTree` 1 `node24`;\          <= HERE IS THE *node* INDICATING CSYS
~ Trail `UI Desktop` `UI Desktop` `PREVIEW_POPUP_TIMER` \
`main_dlg_w1:PHTLeft.AssyTree:<NULL>`;\
~ Activate `main_dlg_cur` `dashInst0.Done`;"
```

I would like to be sure which CSYS I'm using, how can I found the  *node number*? Is there a link with the feat_id or another information available?

Thanks

davidhbigelow, 2019-11-04:
Node Number = FeatureID?

Zepmanbc, 2019-11-04:
Sadly it is not.
But nevermind, i've found how to do, I can select the Csys with the search tool by it's name.
```
~ Command `ProCmdModelImpAppendNew` ;\
~ Trail `UI Desktop` `UI Desktop` `DLG_PREVIEW_POST` `file_open`;\
~ Trail `UI Desktop` `UI Desktop` `PREVIEW_POPUP_TIMER` \
`main_dlg_w1:PHTLeft.AssyTree:<NULL>`;\
~ Trail `UI Desktop` `UI Desktop` `PREVIEW_POPUP_TIMER` \
`file_open:Ph_list.Filelist:<NULL>`;\
~ Select `file_open` `Ph_list.Filelist` 1 \
`my_stl_file_to_import.stl`;\                                   <= HERE IS THE STL FILE NAME
~ Command `ProFileSelPushOpen_Import@context_dlg_open_cmd` ;\
~ Activate `intffeat_file_dlg` `Dlg_PushBtn_Ok`;\
~ Activate `main_dlg_cur` `chkbn.intffeat_reference.0` 1;\
~ Select `intffeat_reference.0.0` `PH.CsysRG` 1 `Csys`;\
~ Command `ProCmdMdlTreeSearch` ;\
~ Update `selspecdlg0` `ExtRulesLayout.ExtBasicNameLayout.BasicNameList` \
`CS0`;~ Activate `selspecdlg0` `EvaluateBtn`;\          <= HERE IS THE NAME *CSO* FOR ME
~ Activate `selspecdlg0` `ApplyBtn`;~ Activate `selspecdlg0` `CancelButton`;\
~ Activate `main_dlg_cur` `dashInst0.Done`;"
```

closed 2019-11-04 by Zepmanbc

davidhbigelow, 2019-11-04:
Ok - Glad you got that figured out!