SimplifiedLogic/creoson #37: Windchill clear_workspace raising error but do the job ¯\_(ツ)_/¯

Zepmanbc, 2019-12-11:
Hi!
I'm using the `windchill` command with `clear_workspace` function to delete 1 file.
Here is the request:
```
{
    'sessionId':'5170641809148701105',
    'command':'windchill',
    'function':'clear_workspace',
    'data':{
        'workspace': 'test_workspace',
        'filenames': ['toto_02.prt']
	}
}
```
the file 'toto_02.prt' is well deleted but it returns me a message error:
```
{status: {
    'error':True,
    'message':'A Pro/TOOLKIT error has occurred: Found.  Check your model/drawing for accuracy.'
    }
}
```

it works fine but I don't understand why there is an error returned.

PS: Do you think it is possible to use wildcard selection for choosing files instead of a list? ( like "toto_*.prt")

adama2000, 2019-12-11:
I'm guessing that Creo is returning a "Found" error to indicate that it found the file -- which is not really an error.  I need to modify the creoson function to ignore any "Found" errors it gets.

Zepmanbc, 2019-12-12:
that would be great, thanks!

adama2000, 2019-12-16:
I'll think about the wildcard option.  That would require looping through all the files in the workspace to find out which ones match, and that would take a while on a large workspace.

adama2000, 2019-12-17:
I've checked in code to ignore the "Found" error on clear_workspace, so the change will be in the next release.

Zepmanbc, 2019-12-17:
great
thank you :pray:

closed 2020-03-08 by adama2000