SimplifiedLogic/creoson #38: Combined views

Zepmanbc, 2020-01-27:
Hello,

is there a way to list and activate combined views?

Thanks

davidhbigelow, 2020-01-28:
I would be surprised if there was a way to do that with the JLINK api....
will check.

But ... can you accomplish that with a Mapkey as an alternative?

Zepmanbc, 2020-01-28:
I can obtain the result of a combined view with views, sections, etc... But i cannot have combined view's list with a mapkey.
I want to get the list of theses views to extract pictures for a visual checking but combined views number and name will change for each file.

DieSwartKat, 2020-01-29:
If you write a mapkey that save the list into a text file you should be able to use a small program to read the text files. This program can then edit a mapkey that will save a picture of each combined view, based on the name.

This mapkey saves a list of all the combine views:

```
mapkey allstates @MAPKEY_NAMEExport AllStates;@MAPKEY_LABELExport AllStates;\
mapkey(continued) ~ Command `ProCmdViewVisTool` ;\
mapkey(continued) ~ Select `visual_dlg0` `RadioSelApplMgr` 1 `combined`;\
mapkey(continued) ~ Select `visual_dlg0` `DisplayCascadeBtn`;\
mapkey(continued) ~ Close `visual_dlg0` `DisplayCascadeBtn`;\
mapkey(continued) ~ Activate `visual_dlg0` `ListZones`;\
mapkey(continued) ~ FocusIn `texttool` `textPH.TextArea`;\
mapkey(continued) ~ Select `texttool` `MenuBar` 1 `FileMenu`;\
mapkey(continued) ~ Activate `texttool` `texttool`;\
mapkey(continued) ~ Close `texttool` `MenuBar`;\
mapkey(continued) ~ Activate `texttool` `SaveAsPushButton`;\
mapkey(continued) ~ Update `file_saveas` `Inputname` \
mapkey(continued) `C:\\PTC\\Bar_Start_Creo\\Drawing-exchange\\allstates.txt`;\
mapkey(continued) ~ Activate `file_saveas` `Inputname`;\
mapkey(continued) ~ FocusIn `texttool` `textPH.TextArea`;\
mapkey(continued) ~ Activate `texttool` `CloseButton`;\
mapkey(continued) ~ Activate `visual_dlg0` `CloseBtn`;
```
note - using CREO 3.0 M050 (Not sure if new versions work the same)
If the above doesn't work simply record the mapkey, save it and edit what you want changed.

You might want to  edit the folder in which to save the text file.

I normally use python, which I can use to read the allstates.txt file get the list of combine views.
In this case, I will get another mapkey together to activate the combined view and then save the view to a jpg (I assume that is what you mean by visual checking).

If you have a list of files that you want to run the above through you could get your code to loop through each file and repeat it over and over.

Note that it is hard to discover errors in mapkeys so if you find you getting stuck, simply run the code to where you see it working and then check the next line till you find your errors.

This is a bit dirty, but I use this all the time to get tasks done that doesn't have JLINK functionality.

Hope I understood your request right.

Zepmanbc, 2020-01-29:
Thank you @DieSwartKat , i will try this 👍

davidhbigelow, 2020-01-29:
It appears that is possible.... just not through JLINK... It appears to be
reserved for ObjectToolkit.... which is a licensed product.

So I would try a mapkey or something to trigger the events you are looking
to accomplish.

Dave

Zepmanbc, 2020-01-30:
Thank you Dave

closed 2020-01-30 by Zepmanbc

DieSwartKat, 2021-07-08:
> Thank you @DieSwartKat , i will try this 👍
> 
> edit: i did not have in mind that logs or other datas were saveable in text files, that's great!

Was just going back to find out if there were indeed a feature in CREOSON for allstates and I come accross my own advice.