SimplifiedLogic/creoson #42: file : assemble - request to add examples

dmtr99, 2020-02-08:
[file : assemble] is quite a cool function, helped me a lot when i needed to place a component in 50 assemblies.

Is it possible to add a few more examples to the provided examples? I would also add a quick description on top of each example.

Requested examples:
- Constraints: on planes (like AFRONT to FRONT with a distance, ARIGHT to RIGHT allined, ATOP to TOP allined)
- Constriants: coordinate system from assembly to coordinate system of component

Is it maybe possible to extract the constraints of one component in an assembly?  Someting like [file : get_constraints] or [component : get_constraints] where you define the assembly and the component to retrieve the constraints.
This way you could use it to paste the component quickly in other assemblies,

The following is a simple example, but used a lot, so I would centainly add it: default placement
```
{
  "sessionId": "~sessionId~",
  "command": "file",
  "function": "assemble",
  "data": {
    "file": "coupling.prt",
    "into_asm": "stub.asm",
    "constraints": [
      {
        "type": "def_placement"
      }
    ]
  }
}
```

davidhbigelow, 2020-02-08:
Thanks!  We use it quite often for many of our automation projects.

We have deliberately kept it *simple* - name references only for CSYS...
they are the most stable thing in Pro/E / Wildfire / Creo.   But requires
automations to be planned for them as known/expected (e.g. skeletons,
interfaces in parts/assemblies) - this assumes a design standard for the
automation / use of models.

Digging deeper into component constraints for assembly has been talked
about for years - but we never seem to agree that it is better than just
using a CSYS as the constraint.   The main reason for this is that we have
seen legacy models that too often were modeled with different
references/constraints relative to new design standards / automation
purposes.  Things modeled years ago - would likely need to be augmented
with the additional features/orientations to be able to be used with
current standards for most companies.  Which then comes to ... do we
require a customer to spend a lot of time
creating/locating/orienting/naming MULTIPLE plane/datum features - or just
create/locate/name a CSYS for the purpose.  CSYS wins everytime in our
book.  So that is what we recommend.

So I don't see this changing - without a REALLY compelling argument.  CSYSs
are very stable and easy to parameterize and use if needed - so we
recommend them as a default.   If the APIs are available in JLINK - we
might consider the defined interface constraint... (but that would need
more research - not sure PTC has exposed that).

We can look at adding a few more examples for the assemble function... I
always think that is a good thing.

Our commercial product (Nitro-CELL
<http://www.simplifiedlogic.com/nitro-cell>) adds some handy options for
assembling things (e.g. wildcard name targets, and even renaming features
with custom incremental names) -- so instead of looping through 50-CSYS
assemble operations - you could simply reduce the call to one that wildcard
hits each CSYS by a name match (e.g. SCREW_*).

   - Assemble Model Wildcard (2:00)
   <https://www.youtube.com/watch?v=sgvuDzc1quM>

This can be mimicked in CREOSON - with a bit of coding...  the net effect
would be the same.

Dave

davidhbigelow, 2020-03-08:
Whopps - did not hit this latest release for better examples... will try to get some the next.

abhimulay, 2025-05-19:
Can you please give an example OR update below code to use Csys as reference?

{
  "sessionId": "~sessionId~",
  "command": "file",
  "function": "assemble",
  "data": {
    "file": "coupling.prt",
    "into_asm": "stub.asm",
    "constraints": [
      {
        "type": "def_placement"
      }
    ]
  }
}

adama2000, 2025-05-26:
There's a couple of examples in the creoson documentation for the file :
assemble function, but here's one from the doc:

{
  "sessionId": "~sessionId~",
  "command": "file",
  "function": "assemble",
  "data": {
    "file": "coupling.prt",
    "into_asm": "stub.asm",
    "constraints": [
      {
        "asmref": "left_asm_ref",
        "compref": "asy_ref",
        "type": "csys"
      }
    ],
    "ref_model": "hose.prt"
  }}

To use this, you first have to determine a CSYS on the assembly you're
assembling to (left_asm_ref in this case) and a CSYS on the component
you're assembling (asy_ref in this case).

If you want to assemble to a CSYS that is on a sub-component rather than on
the main assembly itself, you can do one of two things:
1. Specify the name of the sub-component in the "ref_model" option
(hose.prt in this case).  If there are multiple instances of the same
sub-component, the new component will be assembled to ALL of them.
2. Specify the component path of the sub-component in the "path" option
(not shown here).  This can be useful if there are multiple of the same
sub-component, but you only want to assemble to ONE of them.

Hope that helps,

Adam Andrews

davidhbigelow, 2025-05-27:
> Great! It works. Can you just post the syntax here for component placement
> using datum plane reference. The problem I am facing here are -
>
> constraints=[
> {
> "type": "mate",
> "asmref": "ASM_FRONT",
> "compref": "FRONT",
> "offset": 10.0,
> "flip": False
> },
>
> I am getting error as - RuntimeError: A Pro/TOOLKIT error has occurred:
> General Error.
>
> Not sure whether it is because Mate/Align is now not supported in Creo now.

CSYSs required

adama2000, 2025-05-28:
The problem may be that when your reference is a datum plane, I think you
need to specify which side of the plane you want -- Red or Yellow.

In creoson, you would do that using the "asmdatum" and/or "compdatum"
properties, perhaps like this:

constraints=[
{
"type": "mate",
"asmref": "ASM_FRONT",
"asmdatum": "red",
"compref": "FRONT",
"compdatum": "yellow"
"offset": 10.0,
"flip": False
},

I don't actually have a working example of this, but I believe that's what
you need to do after looking at source code and PTC documentation.

Adam Andrews

abhimulay, 2025-05-31:
[image]

I think just following below steps you may try this case -

Create new blank assembly
Create new blank part > Save if required to retrieve from some path
Try to assemble using default datum plane names.

Thanks!

adama2000, 2025-06-02:
I spent some time trying things this weekend.  I was able to duplicate your
problem.  I have some more things I want to try to figure it out.

Adam Andrews

adama2000, 2025-06-02:
It's possible that creoson won't support this without some code changes,
because the "asmref" and "compref" properties are currently only used when
"type" = "csys".

Adam Andrews

adama2000, 2025-06-02:
> Thanks for trying this out! I think this is a common workflow so looking
> for the fix soon.
>
> Thanks!

Let me clarify -- I duplicated your issue using the PTC API directly,
without even going through creoson. I still need to figure out how to fix
it using the API functions, and only then can we really make a fix to
creoson.

Adam

closed 2026-08-20 by davidhbigelow