SimplifiedLogic/creoson #43: windchill: Logging in windchill automatically

terryxychan, 2020-02-14:
Hi, 

I've been experimenting with Creoson through Creopyson and I haven't found a way to automate certain processes, mainly at authentication. 

Question 1:
- Once I've started Creo using parametic.exe, it prompts me to enter my credentials for my Windchill server. The authentication command "windchill_authenticate" won't go through without manually closing the prompt and switch my Windchill server to offline mode. Is there a way to avoid this or do I have to delete the Windchill server every time I'm done using it and manually set the windchill server everytime I launch up Creo. If so, is there a way to delete and add servers from Creoson?

Question 2: 
- After the server is switched offline, it there a way to reconnect to it? 

Thank you so much!

terryxychan, 2020-02-14:
I have solved these problems by launching creo through creopyson's start command:
```
c = creopyson.Client(ip_adress="localhost",port=9056)
c.connect()
c.start_creo("path to nitro_proe_remote.bat")
```
By using this, Creo will not prompt me asking for credentials. Ans I was able to access my Windchill server after I did:
```
c.windchill_authorize("username", "password")
```

terryxychan, 2020-02-14:
I will close this issue after a week if no one comments.

davidhbigelow, 2020-02-14:
Oh - I was getting ready to take a closer look at that...

Congrats!  Glad you got it working!

Dave

Zepmanbc, 2020-02-16:
I've never tested `winchill_authorize`, I connect to Windchill with SSO, glad to know it works :+1:

closed 2020-03-08 by davidhbigelow