SimplifiedLogic/creoson #44: Symbol replacement

cskn10, 2020-02-26:
Hi, certain symbols needs to be replaced with some other symbols.
The same is applicable for drawing notes also, i was able to do i  that since i got the coordinates of the notes in note.get function, but in symbol.get there is no coordinates, and also there is no symbol.set function to replace is there any other possibility?

adama2000, 2020-02-26:
Looks like PTC does support getting coordinates for Symbols and replacing one symbol with another one -- but we'll have to do some work to make sure it can be used and to get it into creoson.

davidhbigelow, 2020-02-26:
Note -- this may take a while to get around to... (some priorities we are
dealing with at the moment)

If you wanted to get it faster, we might be able to do a bit of consulting
to make sure it fits your needs.

cskn10, 2020-02-29:
Yeah i need it faster, c

> Note -- this may take a while to get around to... (some priorities we are dealing with at the moment) If you wanted to get it faster, we might be able to do a bit of consulting to make sure it fits your needs.
> […](#)
> -- David Bigelow, President Simplified Logic, Inc <https://www.simplifiedlogic.com> | SimilarParts.ai C: 317-431-5454

Yeah i need it faster

davidhbigelow, 2020-02-29:
We looked into it, it is not trivial but can be done.

You can always request consulting to get something prioritized... It is at
the bottom of the CREOSON.com home page.

closed 2026-08-20 by davidhbigelow