SimplifiedLogic/creoson #45: official docker image support

g8gg, 2020-03-03:
consider making a docker image for quick evaluation.

davidhbigelow, 2020-03-08:
The application is self-contained.  Can you elaborate on this request?

terryxychan, 2020-03-13:
If we can launch Creoson as a docker container with mounted volumes to parametric, and port settings, directories etc as environments then it would be more portable? 
I think it might be a great idea

davidhbigelow, 2020-03-13:
I am intrigued by this suggestion - but am confused by the final configuration for such a thing.

Docker w/ Creo Parametric + CREOSON + ???

What about PTC Licensing - does this create a problem?

Graphics?  mapkeys another GUI required operations?

Lot of question on this.   Anyone willing to do a proof of concept?

terryxychan, 2020-03-13:
Can Creoson be in a container but Creo still installed on the host? I'm not sure about the requirements for the communication between Creoson and Creo

adama2000, 2020-03-13:
Creoson needs to be on the same machine as Creo in order to connect to it.
I'm not sure, but it seems to me like container vs host is like being on
different machines.

Adam

davidhbigelow, 2020-03-13:
I am NOT a docker guy...   we have looked at it previously and it seems like a lot of overhead to make work (not for this application - but others we have looked in the past).  It is a good concept and obviously well adopted in the cloud.

Technically YES - it would come down to the security profile on the docker image for ports and any **required mounts to the JLINK libraries for Creo Parametric** ... and whatever port / ip settings would need to be configured so that the Docker could be visible from the host computer/network...   CREOSON is basically a web server with an established connection to Creo.   This requirement would make Creo + CREOSON in a docker seem more logical than just having CREOSON in a docker standalone. (but that brings up other issues - graphics and licensing for Creo being the top of the list).

I noticed this: 
https://thenewstack.io/docker-basics-how-to-share-data-between-a-docker-container-and-host/

Seems like it might be viable - and has many options for implementation/configuration.

I think the use of Docker is really going to make sense if overall architecture is planned for it as a system not just one thing.

An alternative ..... Given the distinct possibility for the need of graphics Creo (and overall performance) - a [Amazon App Stream](https://aws.amazon.com/appstream2/) image with Creo Parametric + CREOSON  might be a more complete and demand-based scalable "vm"....   we have done this in the past for other projects... it works REALLY well if configured correctly.

closed 2026-08-20 by davidhbigelow