SimplifiedLogic/creoson #46: feature: supress

Zepmanbc, 2020-03-04:
Hello,

There is something i don't understand with *feature suppress*

the param `with_children` should suppress all the children of the function.
the param `clip` should suppress everything after the function (whether children or not).

Why the result is an error? With `clip=True` it should suppress everything without arguing?

request:
```
{
    'command': 'feature',
    'data': {
        'clip': True,
        'feat_id': 191,
        'file': 'myfile.prt',
        'with_children': True
    },
    'function': 'suppress',
    'sessionId': '-253955298917346268'
}
```
result:
```
{
    'status': {
        'error': True,
        'message': 'A Pro/TOOLKIT error has occurred: General Error.  Check your model/drawing for accuracy..'
    }
}
```
(feat 191 got children)

If i try to suppress another function, i don't get errors but `with_children` or `clip` params don't change anything on other functions.

davidhbigelow, 2020-03-04:
Hmmm..

The defaults for this function are:

with_children = true
clip = true

Which seems to be what you are requesting... but your expected response does not seem to indicate the request being made... 

I did a similar test with a test model we have:
[image]

Here is the code (sorry JavaScript):

[image]

And here was the resulting output:
[image]

The function seems to work -- can you supply your test model - or more resolution on the test you ar running?   What version / build of Creo?

Dave

davidhbigelow, 2020-03-04:
Oh - I also went and forced the specify of the file in the call...  same result...

[image]

And without forcing the options:
[image]

And as a triple check - just including only the feature_id  (assumes active model)

[image]

One small suggestion (style thing for automating creo) -- naming features might be more handy for automation if you can do that... (helps protect against feat_ids getting messed up when expected in code if your creo model changes with a new feature by a user later).

davidhbigelow, 2020-03-04:
Could there be a relation dependent on these features that is failing when suppressed?

adama2000, 2020-03-04:
With "clip" we just pass that flag on to Creo as an argument to JLink's SuppressOperation class, which is sent to their suppress function.

With "with_children", we use that internally.  If that's set to true, we walk our way down all the child features of the original feature and add them to a total list.  Only at the end do we suppress the whole list of features.

The error you're getting is coming from Creo, not us, so there must be something about the model that's causing the suppress to fail when clip is turned on.

Zepmanbc, 2020-03-05:
Thanks, i'll investigate.

Zepmanbc, 2020-03-05:
I've tried with Creoson 2.4 64bits and it works better than 2.5...

I had to rename the function because feat_id was implemented in 2.5

[image]

`c.feature_suppress(name="EXTRU", with_children=False, clip=False)` => error as expected

`c.feature_suppress(name="EXTRU", with_children=False, clip=True)` => feat 191/220/280 suppressed
[image]
not as expected, 242 is still active

`c.feature_suppress(name="EXTRU", with_children=True, clip=True)` => feat 191/220/280 suppressed, the same as above

`c.feature_suppress(name="EXTRU", with_children=True, clip=False)` => error

it seems that *clip* is required and *with_children* has no incidence.

I use Creo 4.0 M110 on Windows 10

adama2000, 2020-03-05:
After testing with your part, I found a bug in the creoson code where if you specified a feat_id instead of a name, the "clip" flag was NOT properly being passed to Creo, so it was behaving as if clip was always false.  I'll check in a fix for that into creoson.

adama2000, 2020-03-05:
Will that answer the situation or are there still things besides that which aren't working right?

adama2000, 2020-03-05:
I've checked the bug fix in to github

Zepmanbc, 2020-03-05:
Hi Adam,
Thanks for looking at this issue
I don't know if it's a normal way to work but the "with children" in
feature resume seems to not reactivate children functions.
It means that if I want to resume 191 with "with_chldren=True", 220 and 280
are still desactivated

adama2000, 2020-03-05:
... you've uncovered another bug.  I think this bug also explain why you sometimes get errors on the suppress when clip=false but with_children=true.

The bug is that the with_children handling is skipping over unnamed features due to an internal flag setting.  I'll fix that when I do some more testing.

adama2000, 2020-03-05:
I've pushed the new fix to creoson github.

Zepmanbc, 2020-03-06:
Thanks @adama2000 

I've compiled the new version and done some tests:

`c.feature_suppress(name="EXTRU", clip=True, with_children=True)` => supress 191/220/242
`c.feature_suppress(name="EXTRU", clip=True, with_children=False)` => supress 191/220/242
`c.feature_suppress(name="EXTRU", clip=False, with_children=True)` => supress 191/220/242
`c.feature_suppress(name="EXTRU", clip=False, with_children=False)` => error

`c.feature_suppress(name=191, clip=True, with_children=True)` => supress 191/220/242
`c.feature_suppress(name=191, clip=True, with_children=False)` => supress 191/220/242
`c.feature_suppress(name=191, clip=False, with_children=True)` => supress 191/220/242
`c.feature_suppress(name=191, clip=False, with_children=False)` => error

`c.feature_resume(name=191, with_children=True)` => resume all
`c.feature_resume(name=191, with_children=False)` => resume only 191
`c.feature_resume(name="EXTRU", with_children=True)` => resume all
`c.feature_resume(name="EXTRU", with_children=False)` => resume only 191

*feat_id* and *name* do the same now
but I still don't see difference between *with_children* and *clip*
I thought that *clip=True* would have suppress 242, am i wrong?

[image]

adama2000, 2020-03-06:
Here's the JLink doc for what "clip" does: "A Boolean flag that specifies
whether to suppress the selected feature and all the features that follow.
Default value is false."

The question is, what does Creo mean by "the features that follow"?

Feature 242 is not a child of Feature 191, so I wouldn't think it would be
suppressed normally.

Adam

adama2000, 2020-03-08:
With the bug fixes newly released, can this be closed?

Zepmanbc, 2020-03-08:
Yes, thanks
you're right, what is a feature "that follow"...

closed 2020-03-08 by Zepmanbc