SimplifiedLogic/creoson #48: Feature request - Add STL support for interface:export_file

nschoen, 2020-03-25:
It would be really nice to have STL support for file exports through interface:export_file. I use the STL format for point cloud creation by using the vertices of the mesh format and sampling additional points on the surfaces. Currently I use FreeCAD to convert STEP to STL.

Thanks! And great work, creoson simplifies my work a lot.

davidhbigelow, 2020-03-25:
You can use a mapkey for this...  I don’t think off hand there is a
specific export function for this in JLINK - might be wrong.   But a mapkey
will work.

Glad you love CREOSON ... please  brag about it if you don’t mind on the
PTC community -  maybe share an example.  😬

adama2000, 2020-03-25:
The JLink API does appear to support STL export -- it has two different types, "STL ASCII Export" and "STL Binary Export".  So in theory we could add both of those to creoson.

Zepmanbc, 2020-03-26:
You can choose `chord` and `angle` too?

adama2000, 2020-03-26:
The export functions appear to support setting these (whether _creoson_ would support them is another question):

- csysName
- quality (integer 1-10)
- maxChordHeight (double, default=0.1)
- angleControl (double, default=0.1)
- stepSize (double, default=0.0)
- facetControlOptions (various boolean flags)
- sliceExportData (a list of component IDs)

You can specify either quality or a combination of maxChordHeight/angleControl, but not both.

closed 2026-08-20 by davidhbigelow