SimplifiedLogic/creoson #49: interface - export_image : drw => General Error.  Check your model/drawing for accuracy

Zepmanbc, 2020-04-14:
Hello,

I try to export a picture of a drawing (yes a pdf is better but i need a picture for what i'm doing)
I got this issue:

> A Pro/TOOLKIT error has occurred: General Error.  Check your model/drawing for accuracy.

I tried all 4 image types available (JPEG, TIFF, BMP, EPS) without success.
Is it a limitation for drawing files? is it my drawing file the problem (I tried with differents files)?

Second question:
For a drw in the *save as* options there is no JPEG, but there is PNG
Is it possible to add PNG support (even for *prt* and *asm*)?
Thanks

davidhbigelow, 2020-04-14:
The export_image expects a model (part / assembly).

For drawings - I think you would have to run a mapkey to get the png export.

We will have to check to see if there is a PNG export option ... I don't
recall it is a default option.

adama2000, 2021-01-04:
The JLink interface does not support a PNG export, unfortunately.

After testing drawing exports, I added a better error message for any attempt to export an image of a drawing.

closed 2021-01-04 by adama2000