SimplifiedLogic/creoson #51: Model Properties : Accuracy

Zepmanbc, 2020-07-10:
Hi

Is it possible to get current **Accuracy** of a model?
the *value* and if it is *Relative* or *Absolute*

Thank you

adama2000, 2020-07-10:
Yes it would be possible, we could return both relative and absolute accuracy values.  Doing so would be an enhancement.

Maybe it would be a single "get_accuracy" function which took a boolean argument which said whether you wanted absolute or relative accuracy?  Or should it be two separate functions?

adama2000, 2020-07-10:
Actually, maybe it should be a single function which returned both accuracy values.

adama2000, 2020-07-10:
I took a closer look at the Creo documentation -- it says that only one or the other value is given, not both.  So it would be one function that returns a single value plus a relative/absolute flag as you said.

Zepmanbc, 2020-07-10:
That would be perfect the value plus the flag

adama2000, 2020-07-11:
I checked in the file.get_accuracy function.  Appears to work for me, though I couldn't find a model with absolute accuracy for testing.

adama2000, 2020-07-13:
Now tested with an absolute-accuracy model.  Thanks @Zepmanbc

closed 2020-07-14 by adama2000