SimplifiedLogic/creoson #52: Creo 7: Message 'allow_deprecated_config'

Zepmanbc, 2020-10-15:
Hi,

With Creo7 I got this message
[image]

Here is the article: https://www.ptc.com/en/support/article/CS260154

Will you update Creoson to be compatible with Creo>7 versions?

Thanks

davidhbigelow, 2020-10-15:
Next months release 7.0.2.0 Should fix that according to PTC.

Dave

Zepmanbc, 2020-10-15:
OK, i found this with 7.0.2.0 quote
https://www.ptc.com/en/support/article/CS325902?&sourcearticle=CS260154&source=ArticleViewerRelated
Thanks @davidhbigelow

adama2000, 2020-10-15:
I believe we will still need to modify creoson to use a new method for regeneration -- according to this

[image]

There will be a new JLink function to turn the resolve mode on or off.  Creoson will need to use reflection to check whether that function exists, then call it if it does or use the old method if it doesn't.

adama2000, 2021-01-03:
We've added a new function "creo : set_creo_version" which lets you specify what version of Creo you're connecting to.

If you pass it version=7 then it will apply the allow_deprecated_config option mentioned above.  If you're running an older version of Creo, do NOT set it to 7 or it may cause regeneration problems.

This only needs to be called once per creoson connection (calls to connection : connect).

closed 2021-01-03 by adama2000