SimplifiedLogic/creoson #55: Drawing function create_gen_view() requires "scale" parameter

gazontimide, 2020-11-27:
Hi, I am using the python library to control CreoSon.

When using the function create_gen_view(), the documentation states that the "scale" parameter is not required but it will return an error if used without it. If I use the parameter, the function works as intended and the view uses its own scale. I looked at the python's method and it looks fine with the parameter set to scale=None by default.

code : 
view_display_data = {'style': 'no_hidden', 'tangent_style': 'none'}
creopyson.drawing.create_gen_view(clientobj, view='FLAT_VIEW', model_view='TOP', point={'x': 140, 'y': 108}, display_data=view_display_data)

error : 
raise RuntimeError(error_msg)
RuntimeError: No 'scale' parameter given

Thanks,

Sam

adama2000, 2020-11-27:
You're right, that's a bug in the creoson code.  The scale parameter should
be optional.  We'll fix this in the next version.

Thanks for telling us!
Adam

closed 2021-01-03 by adama2000

adama2000, 2021-01-03:
Checked in a fix, it'll be in the next release.

commit 3903d837 (2021-01-03): Fix #55 make scale property optional