SimplifiedLogic/creoson #56: Creo 7: interface/export_3dpdf error

Zepmanbc, 2020-12-03:
Hello
There is an error with this feature in Creo 7, everything is fine with Creo 4

**A Pro/TOOLKIT error has occurred: Invalid Type.  Check your model/drawing for accuracy.**

Here are the logs for an ASM and a PRT
```DEBUG:creopyson.connection:request: {'sessionId': '-3252187263195934486', 'command': 'interface', 'function': 'export_3dpdf', 'data': {'file': 'my_assembly.asm'}}```
```DEBUG:creopyson.connection:response: {'status': {'error': True, 'message': 'A Pro/TOOLKIT error has occurred: Invalid Type.  Check your model/drawing for accuracy.'}}```

```DEBUG:creopyson.connection:request: {'sessionId': '-3813609087025057415', 'command': 'interface', 'function': 'export_3dpdf', 'data': {'file': 'my_part.prt'}}```
```DEBUG:creopyson.connection:response: {'status': {'error': True, 'message': 'A Pro/TOOLKIT error has occurred: Invalid Type.  Check your model/drawing for accuracy.'}}```

davidhbigelow, 2020-12-03:
This might require the special code in the config.pro options.

Zepmanbc, 2020-12-03:
I use the special code, but you means it needs another special code?

adama2000, 2020-12-03:
I'm guessing that one of the settings we set automatically in the PDF
export is no longer valid in Creo 7.  I'll have to test to find out.

Adam

adama2000, 2021-01-04:
I didn't get this error with a test assembly I tried.  Can you send me a test model that causes the issue?

Zepmanbc, 2021-01-25:
Just tried with a simple file and draw, it works.
It must fail because of the size of my files, the problem is that it works on Creo4 with the same files...
I cannot send you theses files...
I will try to investigate by myself

closed 2021-01-25 by Zepmanbc

Zepmanbc, 2021-01-25:
Just found! It is my fault :sweat_smile:
When I launch my script, Creoson automaticaly startup.
But this is the Creoson settings for my main Creo installation (Creo4)
If I launch manualy Creoson with Creo7 settings all is OK

With Creo7 and Creoson settings Creo4, all other functions works well, that is why I did not notice the mistake