SimplifiedLogic/creoson #57: Add support for parameter descriptions

dawnthefawn, 2021-01-13:
Based on what I can see from the JLink documentation, there is support for getting and setting the Description for parameters, but it doesn't look like creoson supports this. This would be a really useful feature for automation.

phaleth, 2021-01-14:
Right. J-Link can do alot with params. Here is an example method.

    /**
     * 
     * @param model
     * @param desiredName
     * @param desiredType
     * @param desiredValue
     * @param desiredDesignation
     * @param desiredDescription
     * @param desiredUnits
     * @return
     * @throws jxthrowable 
     */

    public static Parameter createParamFromScratch(Model model, String desiredName, String desiredType
            , String desiredValue, String desiredDesignation, String desiredDescription, Unit desiredUnits) throws jxthrowable {

        Parameter param = null;
        ParamValue paramValue = null;
        if (desiredType.equals("BOOL")) {
            paramValue = pfcModelItem.CreateBoolParamValue(desiredValue.equals("YES"));
        } else if (desiredType.equals("INTEGER")) {
            paramValue = pfcModelItem.CreateIntParamValue(Conversion.strNumToInt(desiredValue));
        } else if (desiredType.equals("REAL")) {
            paramValue = pfcModelItem.CreateDoubleParamValue(Conversion.strNumToDouble(desiredValue));
        } else if (desiredType.equals("STR")) {
            paramValue = pfcModelItem.CreateStringParamValue(desiredValue);
        }
        
        if (paramValue != null) {
            try {
                if (desiredUnits == null) {
                    param = model.CreateParam(desiredName, paramValue);
                } else {
                    try {
                        param = model.CreateParamWithUnits(desiredName, paramValue, desiredUnits);
                    } catch (jxthrowable x) {
                        param = model.CreateParam(desiredName, paramValue);
                    }
                }
                if(!desiredDescription.isEmpty()) {
                    param.SetDescription(desiredDescription);
                }
                if (desiredDesignation.equals("EQ")) {
                    param.SetIsDesignated(true);
                }
            } catch (jxthrowable e) {
                System.out.println("Parameter " + desiredName + " has not been created");
                System.out.println("jxthrowable:" + "\n" + e);
            }
        }
        
        return param;
        
    }

adama2000, 2021-01-15:
Sounds doable.  I'll work on that.

dawnthefawn, 2021-01-19:
That would be awesome. Thanks!

closed 2021-01-26 by adama2000