SimplifiedLogic/creoson #60: Starting creoson when Creo starts causing "run file 'creoson_run.bat' does not exist"

stekicar, 2021-01-30:
I have placed  CreosonSetup.exe with its full path into parametric.psf as:
```
RUN="C:\CreosonServerWithSetup-2.7.0-win64\CreosonSetup.exe"
```
Creoson starts with Creo but when I click "Start Creoson"button, error "run file 'creoson_run.bat' does not exist" shows up.

If I start Creoson server manually everything works just fine.

What should I do to start Creoson server correctly?

adama2000, 2021-01-31:
I'm guessing the problem is that CreosonSetup needs to be run in the
CreosonSetup folder, so that it can find that .bat file.  When you're
running it from the parametric.psf file, the current working folder is
probably the Creo startup folder.

One solution might be to create your own .bat file that a) does a "cd" to
the C:|CreosonServerWithSetup-2.7.0-win64 directory, b) executes
CreosonSetup.exe, then add a RUN statement to parametric.psf that calls
that .bat file instead.

stekicar, 2021-01-31:
Thank you for reply.
I made new bat file in CreosonServerWithSetup-2.7.0-win64 folder - start_creoson_with_creo.bat:
```
@echo off
cls
taskkill /IM CreosonSetup.exe /F
pushd C:\0_CREO_CONFIGURATION_FILES_OLD\CREO_PYTHON\CreosonServerWithSetup-2.7.0-win64
START C:\0_CREO_CONFIGURATION_FILES_OLD\CREO_PYTHON\CreosonServerWithSetup-2.7.0-win64\CreosonSetup.exe
exit
```
And added this line to the end in parametric.psf:
```
RUN="C:\CreosonServerWithSetup-2.7.0-win64\start_creoson_with_creo.bat"
```
And it works like charm!

Thank you!

closed 2021-01-31 by stekicar

Zepmanbc, 2021-02-02:
If you don't need the JS API and the doc, you can launch only Creoson with `creoson_run.bat` in the Creoson folder
(you need to run CreosonSetup.exe 1 time to create the config files)

adama2000, 2021-02-02:
Very true.  Though one benefit of having CreosonSetup come up is that it
will show you whether the creoson server is already running or not (the
button will say "Stop Creoson" instead of "Start").

Adam

stekicar, 2021-02-02:
Is it possible to start creoson without pressing start button?

adama2000, 2021-02-02:
That's what the creoson_run.bat file is for -- essentially, pressing the
Start button just calls that .bat file and waits 3 seconds to make sure it
started.

adama2000, 2021-02-02:
Note that like CreosonSetup.exe, creoson_run.bat also has to be run from its install directory.

stekicar, 2021-02-02:
Ok. Got it.

stekicar, 2021-02-02:
I just tried to run creoson_run.bat. DOS window pops up and message says "connecting to server, port 9056" and it just sits there. No connection to server is made.

tliebenow, 2021-02-03:
Mine reports:
Starting server, listening on port 9056.

The connection happens in the code you write, I believe.

stekicar, 2021-02-03:
I am getting this message when I try to connect to creoson:
[image]

stekicar, 2021-02-03:
If I run by pressing open button, it works just fine.

adama2000, 2021-02-03:
The "Starting server" message is normal.

All the lines after that are an error that says it couldn't find Creo when
you called the "creo.pwd" function in creoson.  Was Creo running?

Normally you don't see all this, because when you start creoson from
CreosonSetup.exe this DOS window is hidden in the background.

stekicar, 2021-02-03:
I start creoson with "start_creoson_with_creo.bat" so Creo is running. If I change my bat to "START C:\0_CREO_CONFIGURATION_FILES_OLD\CREO_PYTHON\CreosonServerWithSetup-2.7.0-win64\CreosonSetup.exe" everything works without problem.

adama2000, 2021-02-03:
I'll admit I've never used the RUN option in a .psf file.  If you comment that line out and just run your "start_creoson_with_creo.bat" from the command line, does it work then or not?

stekicar, 2021-02-03:
I just tested it. It works ok when run from creoson_run.bat without running it in psf.

adama2000, 2021-02-03:
Okay, so maybe it's a timing issue.  Maybe starting it from the .psf means
it tries to run before Creo has fully finished loading, a problem that
doesn't happen when you have to start creoson manually from the dialog.

stekicar, 2021-02-03:
But that would also be the case when I run creoson.exe from .psf (actually, I start .bat in .pfs file that then starts creoson). And everything is fine. Issue is only when I start creoson thru creoson_run.bat thru my .bat file.

stekicar, 2021-02-03:
I'll try placing sleep 10 second before I run creoson in my .bat file.

stekicar, 2021-02-03:
I put 15 seconds sleep and this is what I get:
[image]
setvars.bat is present in folder.

stekicar, 2021-02-03:
setvar.bat:
```
set PROE_COMMON=C:\Program Files\PTC\Creo 5.0.6.0\Common Files
set PROE_ENV=x86e_win64

set JAVA_HOME=jre
set JSON_PORT=9056
```

adama2000, 2021-02-03:
setvars.bat is created when you run CreosonSetup.exe and fill in the Creo
location and the port number, then press the Start button.  It should
remain there forever after that.

My first thought is that if it can't find setvars.bat, it's because you
haven't done a "cd" to the CreosonSetup directory like you need to.

Zepmanbc, 2021-02-03:
I start creoson from a script, not with Creo because sometimes Creoson crashes and i can detect if Creoson is up or not and if it's down I can reload it in my script

Tips to set Creoson's directory and start Creoson with Python:
>os.chdir("Creoson/path")
>os.startfile(r"Creoson/path/creoson_run.bat")

stekicar, 2021-02-03:
Problem solved. I somehow deleted "p" from pushd when I was testing creoson_run.bat.
pushd C:\0_CREO_CONFIGURATION_FILES_OLD\CREO_PYTHON\CreosonServerWithSetup-2.7.0-win64

I am trying to remove any involvement of Creo user around creoson. This is why I want to start it when Creo starts.
But, I will put in code check if creoson is running and to restart it if it's not,