SimplifiedLogic/creoson #61: creopyson.drawing.get_cur_model raise exception

stekicar, 2021-02-05:
I am trying to get info on current model in drawing:
```
current_model = creopyson.drawing.get_cur_model(c, drawing=None)
print('current_model:', current_model)
```
But I get an error:
```
Traceback (most recent call last):
  File "c:\0_CREO_CONFIGURATION_FILES\CREO_PYTHON\CREO_python_code\drawing_add_approval_symbol.py", line 61, in <module>
    current_model = creopyson.drawing.get_cur_model(c, drawing=None)
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\drawing.py", line 390, in get_cur_model
    return client._creoson_post("drawing", "get_cur_model", data, "drawing")
  File "C:\Users\user\AppData\Roaming\Python\Python39\site-packages\creopyson\connection.py", line 87, in _creoson_post
    raise MissingKey("Missing `{}` in creoson result".format(key_data))
creopyson.exceptions.MissingKey: Missing `drawing` in creoson result
```

Is this a bug?

adama2000, 2021-02-05:
Is there a current model in the drawing?

stekicar, 2021-02-05:
Yes. Its an assembly.

adama2000, 2021-02-05:
This may be a creopyson bug - the return value for "get_cur_model" is a "file", not a "drawing".

adama2000, 2021-02-05:
[image]

stekicar, 2021-02-05:
Ok. Thank you. I was trying to get info for current model in active drawing so I can extract some parameters and use them in symbol creation. I guess  I will have to wait for it to get fixed first.

Zepmanbc, 2021-02-17:
correction in v0.6.2
you can close

closed 2021-03-06 by adama2000