SimplifiedLogic/creoson #62: Request - list_view_details function

gazontimide, 2021-03-16:
Hi, in the next Creoson version release, when using the list_view_details function in a drawing, would it be possible to return even more details about the view? In particular the "View model" of the view?

In our drawings, most of the time we have more than 1 model in the drawing (often family table instances). Knowing which view is from which model would be really helpful for some automation, in my case.

Thanks,

adama2000, 2021-03-16:
We're planning a new release pretty soon, but I should be able to add
that.  Any other details that you want for list_view_details?

gazontimide, 2021-03-16:
I guess "View Type" and "Model Rep" could also be useful, other than, personally, I'm not sure, I always find more ideas as I work with this tool and my needs grow as I go!

adama2000, 2021-03-18:
Looked into this:
- Can add the View Model
- Doesn't look like I can detect View Type, so can't add it
- By Model Rep, do you mean the Simplified Rep?

gazontimide, 2021-03-18:
Yes by Model Rep I meant Simplified Rep. I was foolishly looking at the "Drawing View" information button in Creo and saw very interesting information there, but I didn't look if it was all available in the JLink doc.

[image]

adama2000, 2021-03-19:
Added view_model, simp_rep to output.  It'll be in the upcoming release.

closed 2021-03-19 by adama2000