SimplifiedLogic/creoson #66: Wildcard open all files with same format

DieSwartKat, 2021-04-22:
I've run the CREOSON open command with a wildcard 

data = {'file'  : '*.drw'}

If any of the drawings have an issue opening the command returns an empty list. The specific issues I'm talking about is when a model name in a drawing doesn't exist and the drawings cannot open, until the name issue is fixed.

I have three requests, let me know if I should log them in different tickets.
1. Would it be possible to return a list of the drawings CREO couldn't open?
2. Would it be possible to open the other files which are not broken, instead of just stopping the whole command and return an empty list?
3. It will be ultra amazing if I can have the error message that CREO produce when trying to open the drawing.

CREO 3
CREOSON 2.8.0

thanks

davidhbigelow, 2021-04-23:
Currently no...

Recommended path ... list the files and from drive, list the files in
session...  compare the results/differences.

Alternatively... list the files into memory and loop through them to open
and catch each error.

Dave

DieSwartKat, 2021-04-30:
Thanks Dave, that is what we are currently doing. The wildcard option just speed up the process significantly, if everything goes right.

closed 2021-07-04 by DieSwartKat