SimplifiedLogic/creoson #67: Feature Requests

btxp, 2021-04-30:
Would it be possible to Add a function to replace part or subassembly and return the new filename?
And a function to highlight a part or subassembly in the model tree and the graphics area?

Zepmanbc, 2021-05-01:
It should be possible to do all these actions with mapkeys

DieSwartKat, 2021-07-04:
@btxp you probably solved this already, but I have workflow for extracting mapkeys that can be very handy in your situation. It's worth mentioning in case someone else come across this.

1. In my version of CREO the functionality to create a mapkey is a little hidden so I search for it in the search bar in the top right corner.
[image]
Click the mapkey function as indicated in the picture which will bring up the mapkey dialogue.
2. I use the same mapkey to get the mapkey script from CREO. (a2) which means I select it in the menu and click edit, but if you want you can create a new mapkey.
[image]
If you do select new, you need to fill in a little information before recording, which I trust you will be able to figure out.
3. Once you have the record mapkey window open make sure your model is open and ready at the point you want to start the mapkey from. If you get this wrong, you can always try again. 
4. Select the record keyboard input from PTC Creo Parametric, there is also OS Script function which is handy to run external files.
5. Now you are recording, everything you select will be written down in the record script.
[image]
Once you done, click ok and then save changed. Save the .pro file in a easy place to find. This will only save the mapkey you modified. When you open the newly created .pro file you should see this:

```
mapkey a2 @MAPKEY_NAMEDescription;\
mapkey(continued) @MAPKEY_LABELDescription;\
mapkey(continued) ~ Update `main_dlg_cur` `PHTLeft.simple_search_ph.ss_key_input_panel` `*01*`;\
mapkey(continued) ~ Command `ProCmdSimpleSearch`;

```

I'm using python to edit this text slightly and then get a different result. We built our own python wrapper which I'm using below, but there is a more popular pycreoson library which is well maintained you can also use. 

My python script:

```
from libraries.shared.creoson_connection.CreosonLibrary import  CreosonCommands

creo_ = CreosonCommands()

search_criteria = "*02*"

text = f"""
mapkey(continued) ~ Update `main_dlg_cur` `PHTLeft.simple_search_ph.ss_key_input_panel` `{search_criteria}`;\
mapkey(continued) ~ Command `ProCmdSimpleSearch`;
"""

data = {'script'  : text}

creo_.mapkey(data)
```

The result with the new search criteria:
[image]

The mapkey functionality can often be used to get around functions that are missing from CREOSON, I use it a lot, you can also use it to nest several features together which is often faster than using many CREOSON functions in sequence.

Hope this helps.