SimplifiedLogic/creoson #69: Returning center of gravity as part of return value of file:massprops

iinro, 2021-06-28:
The "mass properties" computed by Creo include items like total mass, volume, several inertia values, center of gravity, etc. Currently, the file:massprops function returns a subset of these values, namely volume, mass, density, surface_area, ctr_grav_inertia_tensor, coord_sys_inertia, and coord_sys_inertia_tensor.
The center of gravity is a very useful parameter of a device, and should be part of the value returned by file:massprops.

adama2000, 2021-06-28:
Should be able to add that, there is a PTC function GetGravityCenter()
which returns "The center of gravity with respect to coordinate frame" as a
3D matrix.

iinro, 2021-06-28:
Sounds perfect. Thank you. I implemented this in a local copy of the code and can open a pull request for your review.

adama2000, 2021-06-28:
Sure, but I won't be able to look at it for a couple of days due to going
out of town.

Adam

adama2000, 2021-10-09:
I was wrong about what it returned, it returns a 3D point, not a 3D matrix.  Good enough?

adama2000, 2022-05-30:
Checked in code, will be released with next release.

closed 2022-05-30 by adama2000