SimplifiedLogic/creoson #71: Delete Feature based on feat_id

DieSwartKat, 2021-07-30:
Hi Guys

I'm making use of the Feature: delete function in CREOSON, but this is only possible for features which have names.  This is the result when I run the Feature List function:
```

[
    {
        "status": "ACTIVE",
        "name": "SIDE",
        "type": "DATUM PLANE",
        "feat_id": 16
    },
    {
        "status": "ACTIVE",
        "name": "TOP",
        "type": "DATUM PLANE",
        "feat_id": 18
    },
    {
        "status": "ACTIVE",
        "name": "FRONT",
        "type": "DATUM PLANE",
        "feat_id": 20
    },
    {
        "status": "ACTIVE",
        "name": "CS0",
        "type": "COORDINATE SYSTEM",
        "feat_id": 30
    },
    {
        "status": "ACTIVE",
        "name": "",
        "type": "WALL SURFACE",
        "feat_id": 548
    }
]
```

I have to renaming feature 548 first before I can delete it as the feature delete function only takes names.

Would it be possible to also accept feature id's in the feature delete function, or have a separate function which accepts feature id's?

davidhbigelow, 2021-07-30:
To encourage better design automation .... we recommend named features.

A workaround would be to record and run a mapkey to perform this
operation.   Swapping the mapkey's ID with your ID just prior to execution.

Dave

DieSwartKat, 2021-07-31:
Thanks David

I agree that this is best practice, and therefore agree with your position. My use case is a little different because I'm trying to delete the FLAT PATTERN feature, which there can only be a single one off. Plus, if I have to delete features by their name, i'm using the rename function first which is using the feat_id to find it, and then delete that feature based on it's name.

Mapkey:
I've recorded the fastest possible mapkey, but it doesn't show the feat id in it, but rather the node position.  I'll therefore have to record it using a search function to find the feature and then delete it base on the selection.

```
mapkey(continued) ~ Select `main_dlg_cur` `PHTLeft.AssyTree` 1 `node13`;\
mapkey(continued) ~ Command `ProCmdEditDelete` ;~ FocusIn `del_sup_msg` `ok`;\
mapkey(continued) ~ Activate `del_sup_msg` `ok`;
```

Would it be an easy addition to add the node position in the feature : list function?

davidhbigelow, 2021-07-31:
Just a hunch ...  what about recording a Mapleton that uses the find dialog
and filter by maybe type?

DieSwartKat, 2021-07-31:
Yes, but that is a lot slower to run. Anyway I'm trying my luck.

Thanks for responding.

closed 2021-07-31 by DieSwartKat