SimplifiedLogic/creoson #72: Edges Detail Requests

a09418849, 2021-08-07:
Is it possible to get the edge's type and sampling points？
Request : 
```JSON
{
  "sessionId": "~sessionId~",
  "command": "geometry",
  "function": "get_edges",
  "data": {
    "file": "TEMP.prt",
    "surface_ids": [
      12
    ],
   "Precision": 2
  }
}
```
Response :
```JSON
........
{
    "edge_id": 35,
    "edge_type":"LINE",
    "length": 10,
    "start": {
      "x": 0,
      "y": 0,
      "z": 0
    },
    "end": {
      "x": 0,
      "y": 0,
      "z": 10
    },
   "sample":[ [0,0,0], [0,0,2], [0,0,4], [0,0,6], [0,0,8], [0,0,10] ]
}
.......
```
Precision could be related in 0~1 or absoluted like above.

```
edge_type : LINE ARC SPLINE.....
```
I think if possible, adding surface type would be perfect. 
```
surface_type: PLANE CYLINDER SPHERE....
```

I need this because I want to get the model section and convert the section to GeoJSON.

adama2000, 2021-08-10:
Hm, well we can get the surface objects that bound the edge, and from there
we could probably get the surface type?

As for the sampling points we may not be able to.  There is a function
called "EvalUV(double)" which "Evaluates the edge at a particular
t-parameter location in terms of the UV coordinates of the edge." according
to PTC.  This returns a structure containing the two UV points, and two
derivatives for the two points (as 2D vectors).  Not sure whether that
helps.

a09418849, 2021-08-10:
In toolkit for C language, I used `ProEdgeXyzdataEval` to get XYZ coord
```c
ProEdgeXyzdataEval(myEdge, edge_param, VtxXYZ, VtxDerv1, VtxDerv2, VtxDir)
```
The VtxXYZ is what I need.
Edge_param is between 0~1. 
edge_param = 0 --> start point
edge_param = 1 --> end point

In PTC User's Guide
```
#include <ProEdge.h>
ProError	ProEdgeXyzdataEval	(
	ProEdge edge	
	/* (In)
	The edge handle
	*/
	double edge_param	
	/* (In)
	The normalized parameter on the edge
	*/
	ProVector xyz_point	
	/* (Out)
	The resulting XYZ point
	*/
	ProVector deriv1	
	/* (Out)
	The first derivative
	*/
	ProVector deriv2	
	/* (Out)
	The second derivative
	*/
	ProVector direction	
	/* (Out)
	The edge direction
	*/
)
```

I don't know is there any function like `ProEdgeXyzdataEval` in Jlink.

About surface type, actually I don't know how to get it by toolkit. But I think it is possible by importing surface id.
```json
..........
"data": {
    "surface_type":"CYLINDER",
    "contourlist": []
    }
..........
```