SimplifiedLogic/creoson #73: Export Program different dimensions?

dmtr99, 2021-08-12:
I tried to use the export program to analyze the dimensions, but I notice that there is a difference between de dimension names.

for example the dimension d21 returned by the export program function should be actually d29

If I check the Program in Creo, it also indicates a different REVNUM:

Creo:
VERSION 4.0
REVNUM 948
LISTING FOR PART DGE_TESTKSDJLKFJS

CreoSon export program:
VERSION 4.0
REVNUM 749
LISTING FOR PART SDFKJSDLKJSDFKL

How can I make sure that the returned Program is correct?

adama2000, 2021-08-12:
In your example it looks like the part numbers are different too -- are you
exporting two different parts?

dmtr99, 2021-08-13:
My excuses, it seems that it was indeed my fault, the function seems to be working perfectly.

And now It works: I created something that automatically modifies the diameter dimension of the holes: It adds the depth of the thread and the amount of holes in the pattern.

example:  d21 goes from M4 to M4 [depthsymbol]8 (4x)

closed 2021-08-13 by dmtr99