SimplifiedLogic/creoson #74: Request: Get selected components update [bom : get_paths]

dmtr99, 2021-08-20:
Is it possible to get the current selected components of the active assembly?

Maybe it is possible to add a return parameter to bom : get_paths if the component was selected?

1011010101, 2022-05-20:
Hello,

Example assembly with three parts (*.prt). Each part corresponds to its own component.

Somehow the "feature:list_selected" function doesn't do what I would expect. For example, it should recognize the component that the user has selected in the tree. The "feature:list_selected" function expects the component (e.g. "C11(name)"). But internally selected is the file (e.g. "name.prt"). You can see this in the search dialog when searching for components. Selected item there is the file. If you remove the selected file and take the component instead, the "feature:list_selected" function finds it.

I think if a file is selected with its own component in it, the "feature:list_selected" function should also list the component.

How can the function find the component selected by the user?

(Creo 6.0.6.0 is used)

I hope you can help me.
Thanks very much.

adama2000, 2022-05-30:
Can you attach a sample model here for us to play with?

1011010101, 2022-07-20:
Please use this example assembly:
[ehs.zip](https://github.com/SimplifiedLogic/creoson/files/9149300/ehs.zip)