SimplifiedLogic/creoson #75: CATIAV5 "Error: Unknown Model Extension"

filoriccardo, 2021-11-04:
Hello to everybody,

I'm trying to open CATIA V5 file (.CATPart, .CATProduct) using creopyson. 

`# creates a connection with creoson server
c = creopyson.Client()

c.connect()
c.creo_set_creo_version(7)
c.server, c.sessionId

# basic Workdirectory handling
c.creo_cd("...\Cover_impermeabile_3.0")
c.creo_pwd(), c.creo_list_dirs()

# selects working directory 
c.creo_cd("demo")
c.creo_list_files()

c.file_open('cover_impermeabile.CATProduct')`

it returns: "Error: Unknown Model Extension"

Is there any solution?

Thanks all

adama2000, 2021-11-04:
That error comes from Creo.  Are you able to do a file-open of that file in
Creo, or do you have to do an import?

Would it work if the file's extension was ".ct" which I think is the normal
Catia file extension?

Adam Andrews

Zepmanbc, 2021-11-04:
CATIA extensions are *.CATPart* (for .prt) and *.CATProduct* (for .asm) but I don't think you can open it like this with Creoson.
You should try with a mapkey.

closed 2023-11-25 by davidhbigelow