SimplifiedLogic/creoson #77: Get Path of active files

DieSwartKat, 2022-04-26:
Hallo Guys

Does anyone know if there is a way to get the path (folder directory in windows) to a part or assembly that is open in CREO at the moment. I'm specifically looking for parts that are not in the working directory, but has been assembled into the model.

I'm assuming the drawing of that part will be in the same folder and therefore want to know the folder so that I can find the drawings.

phaleth, 2022-04-26:
Hi,

Not sure about Creoson, but for J-Link I've found two following examples that are sort of interchangeable.

```java
Window currentWindow = session.GetCurrentWindow();
Model currentModel = currentWindow.GetModel();
ModelDescriptor currModelDescr = currentModel.GetDescr();
// Path without the disk letter, without colon
// and with trailing backslash
String fullPath = currModelDescr.GetPath();
```

```java
Model activeModel = session.GetActiveModel();
ModelDescriptor actModelDescr = activeModel.GetDescr();
// Full path without trailing backslash
String fullPath = actModelDescr.GetDevice() + ":" + actModelDescr.GetPath();
```

adama2000, 2022-05-14:
Yes, the file : get_fileinfo should work, it returns file, dirname, and revision for the requested model.

closed 2022-05-30 by adama2000