SimplifiedLogic/creoson #78: creopyson.feature.list_ does not list features

stekicar, 2022-04-28:
I have the following code:
`import creopyson
from connect_to_creoson_server import connect_to_creoson

conn, c_srv, working_directory, active_file = connect_to_creoson()

if conn:
    lista = creopyson.feature.list_(c_srv, file_=None, name=None, status=None, type_=None, paths=None, no_datum=True, inc_unnamed=None, no_comp=None)
    print("")
    print(lista)
    lista_1 = creopyson.feature.list_params(c_srv, file_=None, name=None, type_=None, no_datum=True, inc_unnamed=None, no_comp=None, param=None, value=None, encoded=None)
    print("")
    print(lista_1)`

Result is like this:
[{'status': 'ACTIVE', 'name': 'MASS_PROP_MAIN', 'type': 'ANALYSIS', 'feat_id': 232}]

[{'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'VOLUME', 'type': 'DOUBLE', 'value': 1978138.021920854, 'encoded': False}, {'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'SURF_AREA', 'type': 'DOUBLE', 'value': 127691.69893617852, 'encoded': False}, {'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'MASS', 'type': 'DOUBLE', 'value': 15.56794623251712, 'encoded': False}, {'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'XCOG', 'type': 'DOUBLE', 'value': 0.39767609722084035, 'encoded': False}, {'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'YCOG', 'type': 'DOUBLE', 'value': 0.493114032625629, 'encoded': False}, {'owner_name': 'MASS_PROP_MAIN', 'owner_type': 'ANALYSIS', 'designate': False, 'owner_id': 232, 'name': 'ZCOG', 'type': 'DOUBLE', 'value': 24.96303083740207, 'encoded': False}]

Thee are no holes listed or any extrusions and they are present in the model.

Is this result as it should be?

adama2000, 2022-05-14:
Sorry for the late reply.

My first guess is that those features are unnamed in your model.  You have "inc_unnamed=None" in your input, and inc_unnamed defaults to false if not specified.  Try "inc_unnamed=True".

stekicar, 2022-05-19:
That was it! Thank you.